micropython/micropython #1688: unix: make minimal

pohmelie, 2015-12-07:
``` bash
$ make minimal
...
CC ../py/objattrtuple.c
In file included from ../py/objtuple.h:29:0,
                 from ../py/objattrtuple.c:27:
../py/obj.h:440:57: error: unknown type name ‘uintptr_t’
     mp_uint_t (*ioctl)(mp_obj_t obj, mp_uint_t request, uintptr_t arg, int *errcode);
                                                         ^
make[1]: *** [build-minimal/py/objattrtuple.o] Error 1
make[1]: Leaving directory `'
make: *** [minimal] Error 2
```

pfalcon, 2015-12-07:
@dpgeorge : So, here it comes again: normal unix build (and coverage) is ok, but minimal is suddenly broken.

dpgeorge, 2015-12-07:
Ok.  So we should add it to the Travis build?

We need a proper solution to this.  We need to automatically build a lot more configurations (eg nan boxing) as well.  There are critical builds (unix, stmhal, cc3200) and the rest are less critical, but still important.  If we build the less-critical ones separately to the Travis run then people are not going to see errors until after the commit goes in, and then there will be some amount of "patching commits".

dpgeorge, 2015-12-07:
But, regarding the issue of building minimal right now: I'd suggest putting #include <stdint.h> in py/misc.h.  stdint.h is C99 so we should have no problem.

pfalcon, 2015-12-07:
@dpgeorge : Sounds good, I wanted to propose similar fix, but decided to ping you, because the main problem is that I don't understand how normal build can be ok, but minimal is broken. General fix works though ;-).

pfalcon, 2015-12-07:
Fixed.

closed 2015-12-07 by pfalcon

dpgeorge, 2015-12-07:
> because the main problem is that I don't understand how normal build can be ok, but minimal is broken

It stumped me too, but I think it's due to this: https://github.com/micropython/micropython/blob/master/py/misc.h#L110 ; so stdint.h is included if unicode is enabled.

pfalcon, 2015-12-08:
Indeed! ;-) Fixed.