micropython/micropython #1690: Unpacking arguments between positional arguments

pohmelie, 2015-12-08:
micropython:
``` python
>>> print("foo", *range(3), "bar")
foo bar 0 1 2
```
cpython 3.4:
``` python
>>> print("foo", *range(3), "bar")
File "<input>", line 1
SyntaxError: only named arguments may follow *expression
```
cpython 3.5:
``` python
>>> print("foo", *range(3), "bar")
foo 0 1 2 bar
```
I think 3.5 behaviour prefered. I don't really know is it hard to implement [pep448](https://www.python.org/dev/peps/pep-0448/), but it is really cool feature.

dpgeorge, 2015-12-08:
So uPy is already half way to pep448 :)  I also don't know how difficult it will be to implement pep448 fully... it may actually simplify the code (due to less restrictions, less error checking), but it remains to be seen.  See issue #1329 for bigger discussion of CPy 3.5 support.

pfalcon, 2015-12-09:
Let's consider this merged into #1329. (My personal opinion is that we can live quite long without this.)

pfalcon, 2016-06-10:
Ok, closing this, I missed the fact that MicroPython gives *incorrect* result.

I now hit this with even simpler case:

~~~
>>> print(*(1, 2), 3)
3 1 2
~~~

Giving incorrect result is definitely not ok.

dpgeorge, 2017-06-14:
This is still not fixed, but in the meantime 1e70fda69fcb4991eb60ed43e610f664ea1319e6 introduces a check for calls like `f(*(1, 2), 3)` and raises a SyntaxError, following CPython 3.4 behaviour (which is what uPy implements).

pohmelie, 2017-06-14:
This is better, than nothing. Maybe issue can be closed.

dpgeorge, 2017-06-14:
Yes let's close it.  The code now follows CPy 3.4, and improving it to follow CPy 3.5+ can be done via #1329.

closed 2017-06-14 by dpgeorge

dbrgn, 2017-06-26:
Just got bitten by this. Support for PEP 448 would be *awesome* :)