micropython/micropython #1693: Internal compiler error on unpacking(?)

pohmelie, 2015-12-08:
``` python
MicroPython v1.5.1-86-g946f870-dirty on 2015-12-08; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> for x in range(*sorted((4, 3))): print(x)
... 
node 159 cannot be compiled
[   1] rule(159) (n=1)
  [   1] rule(119) (n=3)
           id(sorted)
    [   1] rule(135) (n=1)
      [   1] rule(125) (n=1)
        [   1] rule(129) (n=2)
                 int(4)
                 int(3)
    NULL
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
SyntaxError: internal compiler error
```

dpgeorge, 2015-12-08:
A simpler test that fails the same way: `for x in range(*x): pass`.  Changing `range` to anything else doesn't give the error.  So it's most likely something to do with range optimisation.

dpgeorge, 2015-12-08:
Same failure on the following 4 cases:
```python
for x in range(*(1, 2)):
    print(x)
for x in range(**{'end':1}):
    print(x)
for x in range(end=1):
    print(x)
for x in range([] for i in []):
    print(x)
```

dpgeorge, 2015-12-08:
Fixed, see patch above.

closed 2015-12-08 by dpgeorge