micropython/micropython #1697: classmethod fetches parent attribute instead of child attribute when called via child object

mbuesch, 2015-12-09:
CPython 2 and 3 print `1` for the following code, however MicroPython prints `0`

```
class A(object):
        foo = 0

        @classmethod
        def bar(cls):
                print(cls.foo)

class B(A):
        foo = 1

B().bar()
```

I use this to have a generic base class with generic data structures that can be extended by derived classes. The generic classmethod is supposed to use the extended data, if it was called via the derived object.

The actual code doing this is here (generic) https://github.com/mbuesch/awlsim/blob/master/awlsim/core/hardware.py and here (derived) https://github.com/mbuesch/awlsim/blob/master/awlsimhw_debug/main.py

Is there a workaround to this issue?

pfalcon, 2015-12-09:
Do you imply that normal method behavior is different?

mbuesch, 2015-12-09:
Uhm, well I did not try that, because the above is my real use case.
But if I try it (just removed the `@classmethod` line), I see that using normal methods are ok.
So it's just the `classmethod` that has a different behaviour than CPython.

mbuesch, 2015-12-09:
Calling the classmethod via class and not via object works ok. So this might be a workaround for me.

```
class A(object):
        foo = 0

        @classmethod
        def bar(cls):
                print(cls.foo)

class B(A):
        foo = 1

B.bar() #This is ok
B().bar() #This is not
```

pfalcon, 2015-12-09:
> Calling the classmethod via class and not via object works ok.

So, as you can see there're ~ 4 cases: having normal or class method, and calling it via class or object. Someone trying to fix that issue, would need to compare behavior of each (or rely on somebody else to do that). And among those cases, you indeed find a suitable workaround.

mbuesch, 2015-12-09:
I think there are three cases:

```
class A(object):
        foo = 0

        @classmethod
        def bar(cls):
                print(cls.foo)

        def baz(self):
                print(self.foo)

class B(A):
        foo = 1

B.bar()
B().bar() #wrong on upy
B().baz()
```

Only the classmethod-via-object fails (prints 0).

dpgeorge, 2015-12-09:
This patch seems to fix the problem:
```diff
--- a/py/runtime.c
+++ b/py/runtime.c
@@ -934,6 +934,9 @@ void mp_convert_member_lookup(mp_obj_t self, const mp_obj_type_t *type, mp_obj_t
     } else if (MP_OBJ_IS_TYPE(member, &mp_type_classmethod)) {
         // return a bound method, with self being the type of this object
         dest[0] = ((mp_obj_static_class_method_t*)MP_OBJ_TO_PTR(member))->fun;
+        if (self != MP_OBJ_NULL) {
+            type = mp_obj_get_type(self);
+        }
         dest[1] = MP_OBJ_FROM_PTR(type);
     } else if (MP_OBJ_IS_TYPE(member, &mp_type_type)) {
         // Don't try to bind types (even though they're callable)
```

pfalcon, 2015-12-09:
if it seems to fix, and doesn't break testsuite, let's apply it ;-). (But would be nice to have all 3 cases in testsuite.)

dpgeorge, 2015-12-09:
It doesn't break anything existing, so I'll put it in (and add tests).

mbuesch, 2015-12-09:
Yes, this fixes it for me (also in real code). Thanks!
(It now fails in a different spot. Stay tuned.)

dpgeorge, 2015-12-09:
Fixed.

closed 2015-12-09 by dpgeorge