micropython/micropython #1698: uctypes: byte packing not working?

dhylands, 2015-12-09:
I'm trying to pack multibyte data (like UINT16's) on unaligned boundaries.

The docs for LITTLE_ENDIAN say that it uses an packing of 1.
http://docs.micropython.org/en/latest/library/uctypes.html?highlight=uctypes#uctypes.LITTLE_ENDIAN
However, when I tried the following:
```python
unaligned_desc2 = {
    "f1"        : uctypes.UINT8 | 0,
    "arr"       : (uctypes.ARRAY | 1, 2, {
        "s1"      : uctypes.UINT8 | 0,
        "s2"      : uctypes.UINT16 | 1,
                }),
}
data2 = bytearray(b'0123456789')
unaligned2 = uctypes.struct(uctypes.addressof(data2), unaligned_desc2, uctypes.LITTLE_ENDIAN)
print('uctypes.sizeof(unaligned2) =', uctypes.sizeof(unaligned2))
print('unaligned2.f1 =', hex(unaligned2.f1))
print('unaligned2.arr[0].s1 =', hex(unaligned2.arr[0].s1))
print('unaligned2.arr[0].s2 =', hex(unaligned2.arr[0].s2))
print('unaligned2.arr[1].s1 =', hex(unaligned2.arr[1].s1))
print('unaligned2.arr[1].s2 =', hex(unaligned2.arr[1].s2))
```
I get this output:
```
uctypes.sizeof(unaligned2) = 10
unaligned2.f1 = 0x30
unaligned2.arr[0].s1 = 0x31
unaligned2.arr[0].s2 = 0x3332
unaligned2.arr[1].s1 = 0x35
unaligned2.arr[1].s2 = 0x3736
```
I was expecting unaligned2.arr[1].s1 == 0x34 and I was expecting the sizeof the struct to be 7 and not 10.

Is this a bug?
Am I using this incorrectly?
Is the documentation wrong?

pfalcon, 2015-12-09:
> Is this a bug?

Apparently. Finger in the sky, LITTLE_ENDIAN isn't propagated to structural members (like array). Can you start with existing testcases, and make minimal changes to expose an issue?

dhylands, 2015-12-09:
@pfalcon - Sure PR #1699 demonstrates the problem

pfalcon, 2015-12-09:
Ok, so there was wild overlook that structure size depends on its layout type - existing code just blindly assumed LAYOUT_NATIVE. And sizeof() is implemented on descriptor dictionary, so fixing that required quite a bunch of red/green lines to pass a layout type around. d4a874b81e29008713e9de2dd623e62c9f2aa5dc fixes #1699, but @dhylands , keep testing, as there may be more issues (would need testcases for them). I have plans to do bigger refactoring of "uctypes", and can review it firsthand then hopefully.

dhylands, 2015-12-09:
I can confirm that this has fixed  my more general case, which also has the beginning of the array starting on an odd boundary.

closed 2015-12-09 by dhylands