micropython/micropython #1700: Custom exception cannot be re-raised from called function

mbuesch, 2015-12-09:
Consider this code:

```
class MyEx(Exception): pass

def do_raise():
        raise

try:
        raise MyEx()
except MyEx:
        try:
                do_raise()
        except MyEx:
                print("Ok")
        else:
                print("not Ok")
```

On MicroPython it fails with
```
Traceback (most recent call last):
  File "/home/mb/test.py", line 12, in <module>
  File "/home/mb/test.py", line 10, in <module>
  File "/home/mb/test.py", line 4, in do_raise
RuntimeError: No active exception to reraise
```

Things I noticed:
If the raise is done directly in the handler instead of do_raise, it works correctly (prints Ok).
If the exception is `Exception` instead of `MyEx`, it also works correctly.

pfalcon, 2015-12-09:
Whoa! I didn't even think that's possible, always thought that "raise" w/o arguments is lexical-scoped (i.e. must be used in except clause).

Let's start with setting some background first, so give-uppers and losers, authors of Go and Rust, didn't implement exceptions in language at all. Why? Because doing exceptions is *hard*. So instead they told their users: "We're losers, we can't do it. Instead, we burden yourselves with it. Enjoy."

So, uPy isn't like that (and not like PyMite and few other python-alikes), we know that exception are useful, and we have them. But there're limits how much of dirty stuff can be done. And the first rule of doing exception right is "never use implicit exceptions". If you need to catch exception and do something with it (like re-raise), use:

~~~
except MyEx as e:
~~~

We do support form without "as" for compatibility, but its use is strongly discouraged. For example, I don't think it will be ever supported for native/viper codegen.

Beyond that, while it's the most ugliest part of Python, which managed to not die with Python3, but there's sys.exc_info(). Not for writing MicroPython apps - to run some ugly CPython code which use it. You can try to use it if you like, but its use is strongly discouraged.

And looking at the error message you quote, the issue appears that you use another try/except block in existing except: clause. So, simplistic implementation of sys.exc_info() (8b85d14b92a65e92862861038c0fe96e6c616c3e) could just reset any pending exception on start of new try block. So again, the only right solution is get ahold of exact exception object you're going to work with.

mbuesch, 2015-12-09:
>And the first rule of doing exception right is "never use implicit exceptions"

Yes, sure. That's what I what I changed my code to. It probably is cleaner like this anyway.
But I just wanted to point out this difference between MicroPython and CPython that made my program crash.

If it's hard to fix, I wouldn't mind if it wasn't fixed at all.

jonnor, 2024-07-27:
Thank you for this report and the reproducible example. I re-tested this on MicroPython 1.23, and the same behavior is there. So for now I guess we can consider this a known limitation.