micropython/micropython #1701: Recasting current baremetal modules to "umachine"

pfalcon, 2015-12-09:
@dpgeorge , @dhylands : I mentioned this couple of times in various related comments, but didn't get response, so here's explicit question:

I'd like to have more or less complete implementation of "machine" module for unix port. The primary motivation for that is testing, but we'll see where it goes. And good part of module would be implemented in Python using FFI, while some part in C, with requires separating module names. How we handle that so far is having builtin module with "u" prefix, and "module symlinks" for baremetal ports. So, I'm asking you how we should handle this for "machine" module.

dpgeorge, 2015-12-11:
On unix, having a "umachine" module as the builtin C implementation, plus "machine.py" for extra stuff sounds good, just like we usually do it.

On baremetal, I would expect these ports to provide a full implementation of machine module straight away.  But I can see how there might be cases where it's not fully implemented, or you want to override it (perhaps to fix a bad implementation).  So going the "umachine" + symlink route sounds good here as well, since we use this pattern for pretty much all modules on baremetal.

pfalcon, 2015-12-16:
@dpgeorge : Thanks

@danicampora : Any comments? Would you follow with this change? If so, should cc3200 port just be changed along the other ports?

danicampora, 2015-12-16:
Yes, I'm OK with this change.

pfalcon, 2015-12-18:
Implemented in 2eb844e0b4e649034115dd6ca77dcdecd92ef547

commit 2eb844e0 (2015-12-18): ports: Rename "machine" module to "umachine".

To let unix port implement "machine" functionality on Python level, and
keep consistent naming in other ports (baremetal ports will use magic
module "symlinking" to still load it on "import machine").

Fixes #1701.
closed 2015-12-18 by pfalcon