micropython/micropython #1707: Nested arguments in str.format

pohmelie, 2015-12-11:
micropython:
``` python
>>> str.format("{text:{align}{width}}", text="foo", align="^", width=20)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid format specifier
```
cpython 3.4.3:
``` python
>>> str.format("{text:{align}{width}}", text="foo", align="^", width=20)
'        foo         '
```
I read  #407 and #574 (as wiki said). I understand, that core contributors probably don't like `str.format` and `str.format` not fully supported, but I prefer this method of string formatting and want this to be implemented.

pfalcon, 2015-12-11:
> and want this to be implemented.

Patches welcome (don't forget to include code size increase for stmhal port) ;-)

Otherwise, duplicate of https://github.com/micropython/micropython/issues/812

closed 2015-12-11 by pfalcon

pohmelie, 2015-12-11:
This is not the reason to close issue.

pfalcon, 2015-12-11:
Duplicates *are* reason to close issues, otherwise there will be dozens of reports for the same issue and it will be impossible to use the bugtracker. Please always search for existing report before opening a new one.

pohmelie, 2015-12-11:
@pfalcon, oh I missed it is opened. My fault.