micropython/micropython #1710: bytearray slice assignment fails for derived types

mbuesch, 2015-12-11:
The type check for slice assignment to bytearray derived types seems to be a bit too strict:
```
class MyBytearray(bytearray): pass

a = MyBytearray(4)
b = MyBytearray(4)
a[0:4] = b
```
results in this error:
```
Traceback (most recent call last):
  File "test.py", line 5, in <module>
NotImplementedError: array/bytes required on right side
```

dpgeorge, 2015-12-11:
It requires a little bit of work to support derived types on rhs of assignment.

Work around: use `b[0:4]` or `memoryview(b)` on RHS.

pfalcon, 2015-12-11:
This issue happens not just for "derived" types, but also for "compatible", like assigning from array of different element types. So, we have "fastest path now" (effectively, a memcpy), and what we should do is to have "slowest path" - assigning from arbitrary iterable. Of course, nobody is interested to spend time on "slowest path", that's why it's not implemented ;-).

mbuesch, 2015-12-11:
Ok, I see. I'll have to check, whether I can do a workaround in a performance neutral way, because this affects the most performance critical part of my software.

dpgeorge, 2015-12-11:
Using a memoryview (pre created perhaps) should not impact performance at all (or very little).

projectgus, 2024-08-28:
Labelling as `docs` as this has since been partially documented in 86c7507233929366ff17c9666200a33123491a8e but the description could use some elaboration.