micropython/micropython #1711: list slicing issue on reversed slicing

mbuesch, 2015-12-12:
Reversed list slicing doesn't work correctly, if indices are specified.

Micropython:
```
>>> [1,2,3,4,5][:0:-1]
[]
>>> [1,2,3,4,5][:1:-1]
[1]
>>> [1,2,3,4,5][:2:-1]
[2, 1]

>>> [1,2,3,4,5][0::-1]
[5, 4, 3, 2, 1]
>>> [1,2,3,4,5][1::-1]
[5, 4, 3, 2]
>>> [1,2,3,4,5][2::-1]
[5, 4, 3]
```

CPython 3.4:
```
>>> [1,2,3,4,5][:0:-1]
[5, 4, 3, 2]
>>> [1,2,3,4,5][:1:-1]                                                                                        
[5, 4, 3]
>>> [1,2,3,4,5][:2:-1]                                                                                        
[5, 4]

>>> [1,2,3,4,5][0::-1]
[1]
>>> [1,2,3,4,5][1::-1]
[2, 1]
>>> [1,2,3,4,5][2::-1]
[3, 2, 1]
```

pfalcon, 2016-01-03:
@mbuesch: As you can see, unlike other issues you reported, hat didn't catch immediate attention from other developers, so please take an invitation/encouragement to look into this yourself, if you're interested. This should be right good task for initial acquaintance with codebase. Actually, you may start with looking at #1481.

closed 2016-10-30 by pfalcon

pfalcon, 2016-10-30:
https://github.com/micropython/micropython/issues/2577 claims to have fixed these issues.

PR #2577 (2016-10-30): Fix reverse slicing of lists.
This fixes the slicing when using reversed lists (negative step). Closes #1711.