micropython/micropython #1721: downloadable docs out of date

markhellewell, 2015-12-15:
Hi!  I just went to download the HTML docs for 1.5.1 and noticed they’re missing:

https://readthedocs.org/projects/micropython/downloads/

Cheers!

pfalcon, 2016-01-03:
There was known issue with doc builds at readthedocs.org (where you link). It is fixed now. Only "latest" version is available now. @dpgeorge : Perhaps tag 1.5.2.1 to get downloadable docs into peoples' hands?

closed 2016-01-03 by pfalcon

dpgeorge, 2016-01-03:
> There was known issue with doc builds at readthedocs.org (where you link). It is fixed now.

Thanks!  I didn't realise this has been broken for a while.

> Perhaps tag 1.5.2.1 to get downloadable docs into peoples' hands?

I'd rather not confuse the version numbers by doing this... there is a "latest PDF" link (along with latest HTMLZip) on that dl page, as well as links to these on the readthedocs pop up menu, so people can get the downloadable docs.  And hopefully v1.6 is not far away :)

pfalcon, 2016-01-03:
> here is a "latest PDF" link (along with latest HTMLZip) on that dl page

Yes, at the bottom of it. Ok, let's hope it'll fix itself over time.