micropython/micropython #1734: ESP8266 socket stacks

lamazavr, 2015-12-20:
Hi, I am trying to implement NTP request via esp sockets.
So, I connect to Wifi and do:
>>> s = esp.socket()
>>> def socket_printer(socket, data):
...     print(data)
>>> s.onrecv(socket_printer)
>>> NTP_QUERY = b'\x1b' + 47 * '\0'
>>> s.connect(('88.147.254.235', 123))
>>> s.send(NTP_QUERY)
48
>>> 

After this I can`t type any symbol. 
Maybe I do something wrong?

dpgeorge, 2015-12-22:
> After this I can`t type any symbol.
> Maybe I do something wrong?

You might need to wait for `s.connect` to suceed before doing `s.send`.  You can set a callback using `s.onconnect`.

This callback event model is not very fun to use.

dpgeorge, 2016-01-30:
This should be fixed when #1806 is closed.

pfalcon, 2016-03-24:
esp8266 switched to lwIP sockets.

closed 2016-03-24 by pfalcon