micropython/micropython #1736: uos.dupterm() behavior in underspecified, implementations broken

pfalcon, 2015-12-20:
While not directly related to HW API, https://github.com/micropython/micropython/wiki/Hardware-API#os-module-additions specifies os.dupterm(), used for:

> os.dupterm([stream_obj]) get or set duplication of controlling terminal, so that REPL can be redirected to UART or other (note: could allow to duplicate to multiple stream objs but that's arguably overkill and could anyway be done in Python by making a stream multiplexer)

Unfortunately, this description is not clear/detailed enough, or alternatively, current implementations of it in stmhal and cc3200 ports are not general or just broken. First of all, it should be made clear if the talk is about duplicating "controlling terminal", or replacing it. If there's talk about duplicating, then there should be immediate talk about non-blocking streams. Otherwise this line https://github.com/micropython/micropython/blob/master/cc3200/hal/cc3200_hal.c#L187 will simply hang the entire REPL, as it will wait until a character is available on that particular stream, ignoring all other input sources.

And then if there's requirement to non-blocking streams, it should be kept in mind that all other REPL sources should be non-blocking too, otherwise standard source can block custom installed one. But requirement of non-blocking sources is merely a minimal condition to get it all working, but working quite inefficiently and requiring weird workaround to not make it burn CPU cycles (and a battery of battery-powered device), e.g. https://github.com/micropython/micropython/blob/master/cc3200/hal/cc3200_hal.c#L196. So, next step is making requirement that REPL sources are pollable, and actually poll them.

That's quite a bunch of requirements to get it working in the general case, sure.

dpgeorge, 2016-01-05:
> So, next step is making requirement that REPL sources are pollable, and actually poll them.

Yes I think this is the correct approach, to use select.  Select/poll is implemented on pyboard for UART, USB, CAN and sockets, and should be extended to other stream-like objects (eg I2C, SPI).  CC3200 uses stmhal's moduselect.c source so it (in principle) also has the same functionality.  We should make this functionality more generally available for all ports.

pfalcon, 2016-01-05:
> We should make this functionality more generally available for all ports.

And start with unix port, and implement ability to poll native POSIX and custom objects together. Until that is done, the above doesn't seem like viable approach.

There's alternative of using "push" design, specifically for REPL dupterm's, where each REPL data producer pushes its data to a common queue, from which REPL (single consumer) fetches it. But that again doesn't map easily to "pull" POSIX model (a natural implementation for it would be to run each produces in its own thread, but I don't think we want to add threading requirement for that). And polling for custom objects on unix port would still need to be implemented independently.

Overall, there's following dichotomy: POSIX forces pull model, and poll, and some ports are ok to follow. Still, for many baremetal/minimal-RTOS cases, the most efficient solution is push-based (IRQ handler pushes data). The "ideal" solution would be to have REPL push-based, and dupterm handling either pull or push based. Then a particular port could mix and match whatever is best suitable for it.

pfalcon, 2016-01-05:
> We should make this functionality more generally available for all ports.

And to be more specific about steps towards using that consistently, how about dropping adhoc UART .any() method? ;-)

dpgeorge, 2016-01-05:
> Still, for many baremetal/minimal-RTOS cases, the most efficient solution is push-based (IRQ handler pushes data).

The push here is really just a signal that data is ready, and can be used to implement select/poll (IRQ sets a simple flag).  The top level event handler is still going to be a loop that checks all these flags, so easiest thing is to just keep everything pull based with IRQ signals.

> And to be more specific about steps towards using that consistently, how about dropping adhoc UART .any() method? ;-)

uart.any() is a simple and efficient way to check for data, instead of allocating memory and doing many calls to select/poll.  Select/poll can be implemented in terms of uart.any() [and so we could make a requirement that all objects implement any() at the minimum, and then they just work with select/poll].

danicampora, 2016-01-05:
> And then if there's requirement to non-blocking streams, it should be kept in mind that all other REPL sources should be non-blocking too

@pfalcon, being non-blocking is an inplicit requirement. As @dpgeorge says we should add as a requiremente that the .any() method must be implemented too.

pfalcon, 2016-01-05:
> The push here is really just a signal that data is ready, and can be used to implement select/poll (IRQ sets a simple flag).

No, the push is real data. It can be used to implement extra hierarchical processing, the point is that it can push real data into REPL. Well, maybe not straight from IRQ handler. But simplest form of RTOS functionality is still where tasks are represented by event handlers, i.e. it's still push processing (IRQ pushes to queue, then RTOS scheduler pushes it to task). And any resource-efficient, cooperative OS would be like that, because again, anything else is hierarchies and bloat. Examples are: Contiki, ESP8266's ROM RTOS, BlueCore RTOS (https://github.com/pfalcon/blutunode), Windows16, etc.

So, the question boils down to whether we can standardize on the "lowest denominator" REPL implementation, which is push based. Currently, we have 2 implementations, with push-style undermaintained, and growing behind pull-based. But if we drop push-based, we cut off pretty interesting subset of RTOSes.

pfalcon, 2016-01-05:
> uart.any() is a simple and efficient way to check for data

In other words, you're not ready to switch to single consistent way to deal with waiting for data readiness? ;-) Because of efficiency? Then why wanting to force push-based designed to be pull-based, if it's more complex and less efficient (and for large subsets of push-based systems, they simply can't work in general pull-based manner (there's no persistent stack to keep local state, which most pull-based designs take for granted)).

> Select/poll can be implemented in terms of uart.any()

This is rather inefficient implementation. Instead, IRQ handler should put an object on "ready" list, to avoid list full scans. So, .any is orthogonal to efficient multiple object polling implementation, not elementary feature in terms of which it can be implemented.