micropython/micropython #1739: Compiler replace const values in wrong places

Anton-2, 2015-12-21:
```python
import uctypes
CST = const(0)
test = uctypes.struct(uctypes.addressof('ABCD'), {'CST' : uctypes.UINT32})
print(test.CST)
```
will get ```KeyError: <value>```

and 

```python
class A:
    CST = 10
```
brings ```SyntaxError: can't assign to literal```

In both cases, the compiler is replacing CST with the const value, making the expression meaningless.

dpgeorge, 2015-12-22:
Yes, I confirm this.  The first issue (test.CST) is a bug.  The second issue (CST=10) is a "feature" because it stops you from reassigning constants and confusing locals with globals.

dpgeorge, 2015-12-23:
Also related and needs fixing (currently fails an assertion on unix port):
```python
class A:
    C1 = const(1)
    def x(self):
        print(A.C1)
    def y(self):
        print(self.C1)
A().x()
A().y()
```

dpgeorge, 2016-11-22:
This is now fixed.  See 6d310a5552f43b3f984a5d8020b41301ab7937a4.

closed 2016-11-22 by dpgeorge

pmp-p, 2020-08-04:
> The second issue (CST=10) is a "feature" because it stops you from reassigning constants and confusing locals with globals.

This should be kept open until it **warns** on re-assign (before first expansion ofc) instead of stopping. 

Actual behaviour is plain wrong for a code intended to be shared on multiple platforms and interpreters with compile time choices that should not require a preprocessor.
 eg:
```
if __import__('sys').implementation.name != 'micropython':
    exec('__import__("builtins").const = lambda x:x',{},{})

some64test = 0

if some64test:
    pt_size = const(8)
else:
    pt_size = const(4)

print(  pt_size )
```

or simply keep actual behaviour but then accept an immediate eval() inside const( ... )  as a useable workaround
 eg assuming pt_size has been set before the import :
```
pt_size = const( eval("%s"% __import__("__main__").pt_size) )
```

peterhinch, 2020-08-04:
The issue seems to be a class variable having the same name as a global constant. This
```python
CST = 0

class A:
    CST = 10

print(CST, A.CST)
```
produces 0, 10 es expected. Making the global **CST** a **const** produces an exception
```python
CST = const(0)

class A:
    CST = 10
```
produces **SyntaxError: can't assign to expression**
```
MicroPython v1.12-649-g27767aafa-dirty on 2020-07-21; linux version
```