micropython/micropython #1741: boot.py should be site.py

pfalcon, 2015-12-22:
Python has a standard module which gets imported during interpreter startup: "site" https://docs.python.org/3/library/site.html . We should consider how much boot.py semantics differ from site.py, add support for site.py, and ideally deprecate boot.py.

This issue comes up after considering how to add Python-level features to unix port (e.g. "help()"), but also how to clean it up for other new ports.

danicampora, 2015-12-22:
Interesting point.

dpgeorge, 2015-12-26:
I really like boot.py.  It's obvious what it does, makes sense from a microcontroller point of view, and can be kept minimal (it doesn't need to contain anything, or even exist, for a successful boot).

site.py is actually a predefined module, part of the stdlib, not something configurable by the user.  Users should create a usercustomize (or sitecustomize) module to do their own config on start-up of the interpreter.  But this is too much bloat.

So I'd say the semantics are too different and boot.py should stay.

dhylands, 2015-12-26:
Presumably, if someone really wanted site.py support then boot.py could just import it.

I think there is definitely merit in having some type of startup file for unix micrpython. Is site.py imported or execfile'd?

I really like the fact that boot.py is execfile'd and can modify the defaiult REPL environment.

pfalcon, 2015-12-27:
> site.py is actually a predefined module, part of the stdlib, not something configurable by the user. Users should create a usercustomize (or sitecustomize) module to do their own config on start-up of the interpreter.

Well, all that is CPython's paraphernalia. Dry residue is that Python (the language) imports module "site" on startup. It can of course be empty, and then it can be only implementation-specific behavior how lack of such module is handled, the obvious choice is to handle it in the same way as empty module.

pfalcon, 2015-12-27:
> Presumably, if someone really wanted site.py support then boot.py could just import it.

The logic is opposite - if someone really wants to have non-standard name, then it easily can be done from standard name.

> I really like the fact that boot.py is execfile'd and can modify the defaiult REPL environment.

Python (the language) prefers imports, as it automagically answers questions like "what happens if module executed twice". Python (the language) allows to modify "REPL environment", but using module import approach, guides users into doing that in reliable manner (e.g., by modifying builtins module, to ensure these changes are available across modules).

Though if the idea of boot.py is not to leave entry in sys.modules, then maybe it indeed should exist in addition to standard site.py machinery (not leaving entry in sys.modules can be easily emulated by del'ing it from there, though if the idea how all this will behave on 2KB heaps then yes, not even adding it makes sense).

dhylands, 2015-12-27:
> Presumably, if someone really wanted site.py support then boot.py could just import it.
> The logic is opposite - if someone really wants to have non-standard name, then it easily can be done from standard name.

I don't see how that could work. How can I can one do import site and have it leave a symbol like foo in the REPL environment (i.e. if you do dir() at the REPL and see foo listed)?

That can easily be made to work because boot.py is execfile'd.

pfalcon, 2015-12-27:
~~~
>>> import builtins
>>> builtins.foo = lambda: 1
>>> foo()
1
~~~

Once again - this will work even if you try to access foo() from another module.

dhylands, 2015-12-27:
Ok - so we can do it both ways:

Method 1:
- have the code execfile boot.py
- have boot.py import site.py
- have boot.py define any customizations (to be available at the REPL)

Method 2:
- have the code import site.py
- have site.py define any customizations (to be available at the REPL)
- have site.py add monkey patch the customizations into builtins

I think that both methods should use approx the same amount of flash.

By using your arguments of bloat, Method 1 seems like a clear winner (uses less RAM). With method 2, you're forced to import a module you may or may not use, and then you have to waste more RAM getting the customizations monkey patched in.

pfalcon, 2015-12-27:
There's another argument - what language we implement. If Python, then there's no boot.py. And I myself implemented many features which are subset or variation of Python functionality. But was doing that with a specific aim - to have an efficient core, and implement less and less variations going further. Less, not even more.

In this regard, unix port neither has support boot.py nor there's much sense to support it (it's not Python). So, I submitted this ticket to see if there's interest to move baremetal to more common variant (I don't propose to drop boot.py support right away, but using common code which try to import configurable list of modules make sense). If there's no such interest, well, we'll have 2 pieces of code which do almost same thing in 2 different ways - one in Python standard way, and one - not.

rolandvs, 2016-01-24:
Why not keep the `boot.py`, but have it named `.boot.py` so it will not show up in the directory? Also a neat way to hide some other modules from showing up... ;-)

pfalcon, 2016-03-31:
@ncoghlan : I'm curious what you might say on this matter - we need opinions from upstream folks, or risk being stuck in feudalism on this stuff ;-).

ncoghlan, 2016-04-01:
The two questions seem somewhat orthogonal to me. The implied "boot.py" for CPython is the operating system's process initialisation code, and all the related bootstrapping that CPython does to get itself up and running:  https://www.python.org/dev/peps/pep-0432/#the-status-quo

When working on a Python implementation, it's important to keep in mind that a lot of the stuff under https://docs.python.org/3/using/ is actually CPython specific - other implementations are going to be different in terms of what they support based on the needs and interests of their particular audience. In MicroPython's case, one of the key differences is to avoid assuming there's an operating system around to help out.

That said, having a predefined boot.py for the UNIX port that strives to emulate CPython's command line handling, virtual environment handling and other startup behaviours could potentially make a lot of sense - that would make it relatively straightforward to isolate the parts that are only relevant when running in the context of a full operating system from those that are also applicable when running as part of an embedded system without a separate OS.

ncoghlan, 2016-04-01:
As far as the specific suggestions of letting `import site` and `python -m site` do arbitrary things go, no, that API and CLI is defined in the library reference and aren't a CPython implementation detail: https://docs.python.org/3/library/site.html

The specifics of the `sys.path` manipulation and some of the other side effects do vary by implementation, but it isn't a complete free-for-all.

pfalcon, 2016-04-01:
@ncoghlan : Thanks for the detailed reply. I actually meant to ask about something different/more specific: "Currently, some ports of MicroPython use boot.py as a script/module automatically imported during interpreter startup. The closest thing CPython has is site.py module (details of semantics of boot.py and site.py importing differ a bit, but if one says they're minor details, one won't be incorrect). So, as an upstream developer (and without trying to wear MicroPython developer shoes), would you: 1) recommend abandoning adhoc module naming and adhoc semantic differences, and just follow CPython semantics of importing site.py; or 2) agree that every Python implementation may treat that matter on its whim".

(It's formulated with obvious connotations, and that's the whole point - we have a lot of disputes how to implement things not defined by Python (with CPython being closest thing to Python); we usually didn't have much disputes how to implement things defined by Python. By getting indulgence to randomly stray away from doing things in Python way *without very good reason*, we risk losing direction of coherent development and instead being swamped with disputes.)

ncoghlan, 2016-04-01:
I'd suggest that `boot.py` is closer in semantics to the `PYTHONSTARTUP` environment variable than it is to `site.py`, only using a specific name by convention rather than an environment variable: https://docs.python.org/3/using/cmdline.html#envvar-PYTHONSTARTUP

I do think it would be good for the *nix port of MicroPython to support the general semantics of site.py (including the `-s` and `-S` switches to control which parts of it get executed, the notions of user specific imports and virtual environments, and execution via `-m` to inspect the resulting settings), but I think those are all markedly less important for the embedded use case (where it's more reasonable to assume total control over the system than it is in a full *nix environment)

pfalcon, 2016-04-01:
> I'd suggest that boot.py is closer in semantics to the PYTHONSTARTUP environment variable than it is to site.py

Thanks, that gives strong argument to proponents of boot.py, and I'm ready to give up on site.py idea unless they themselves decided to switch to site.py in the meantime ;-).

> I do think it would be good for the *nix port of MicroPython to support the general semantics of site.py 

Besides MicroPython's affinity towards simplicity, there's currently no users for these advanced features. As uPy progresses further, they may appear, but in the meantime, the issue at hand is supporting auto-loading of site.py at all.

dpgeorge, 2016-04-04:
> unless they themselves decided to switch to site.py in the meantime

Nope, I didn't :)

closed 2016-05-28 by pfalcon

pfalcon, 2018-12-15:
> I'd suggest that boot.py is closer in semantics to the PYTHONSTARTUP environment variable than it is to site.py

From `python3 --help`:

> PYTHONSTARTUP: file executed on interactive startup (no default)

So, claim of similarity of boot.py and PYTHONSTARTUP is rather far-fetched.

So, this issue is still pertinent: for no truly good reason, MicroPython went into inconsistency with CPython regarding how the load-on-startup file is named (boot.py vs site.py) and details of loading process (exec'ing vs import).