micropython/micropython #1742: Ctrl+C on non-empty line acts like Enter instead of Cancel/Escape

pfalcon, 2015-12-22:
As reported by @pohmelie at https://github.com/micropython/micropython/pull/1733#issuecomment-166243462 :

> Also, if I try ctrl-c in repl it looks like it actually do enter, even on unix build.
~~~
$ ./micropython
MicroPython v1.4.5-654-g37f414f on 2015-12-21; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> 
>>> 
>>> fooTraceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'foo' is not defined
>>> 
~~~

dpgeorge, 2015-12-22:
Ok, this particular issue fixed in e9751d2ac057070c52a7a67bd6e8b97e944ee088.  [stmhal is ok, since lib/utils/pyexec already had the check for ctrl-c.]

Note that not printing "KeyboardInterrupt" on ctrl-C at the REPL (as per CPython) is by design (rightly or wrongly).  The REPL is not executed within an nlr protection block and so shouldn't raise any exceptions (only executed code is nlr protected).

pfalcon, 2015-12-22:
Thanks for quick fix!

> Note that not printing "KeyboardInterrupt" on ctrl-C at the REPL (as per CPython) is by design (rightly or wrongly).

Yep, I figure that, that's why I didn't raise that as an issue, and as nobody else did either, I hope everyone agrees it makes sense. For me, not printing extra line is aesthetically cleaner, kind of like bash handles ctrl+c.

closed 2015-12-22 by pfalcon