micropython/micropython #1743: uctypes: assign to scalar in an array is not fully implemented

Anton-2, 2015-12-23:
```python
import uctypes

desc = {
    # arr is array at offset 0, of UINT32 elements, array size is 2
    "arr": (uctypes.ARRAY | 0, uctypes.UINT32 | 2),
}

data = bytearray(8)
s = uctypes.struct(uctypes.addressof(data), desc)

s.arr[0] = 0x11223344
```

gives ```TypeError: 'struct' object does not support item assignment```

It only works for UINT8 (because the array comes out as a bytearray).

I've got a patch+test for this, but #1698 mention I big refactor coming.
Are you interested in it anyway ?

pfalcon, 2015-12-23:
Yes, and indeed, would be nice to get this in before I start on that refactor.

Anton-2, 2015-12-23:
Ok, please have à look at #1745

Anton-2, 2016-01-11:
Implemented in 26ed00118b48ccf9fca1308015e269f1fc400a79

closed 2016-01-11 by Anton-2