micropython/micropython #1746: How to construct a python object pointer given an address

dhylands, 2015-12-23:
If I know that there is a python object at a particular address (say by looking at pyb.info(1) output), is there a way to construct a variable which points at that memory so that you can query the object (i.e. determine its type, etc)?

I know you can use uctypes.addressof

In C, given an address addr, I could just do a cast (mp_obj_t)addr

I think that it would be useful to have such a function. Any thoughts on which would be the most appropriate module to put this in?

On a related note, I also think it would be useful to have some variations of gc_dump_alloc_table. One would be just to dump a more limited range (rather then the entire heap) and another would be to actually dump the objects in a range of heap. The latter could be quite nicely done using some type of iterator/generator.

dpgeorge, 2015-12-23:
I'd guess that the 'O' type of array can be used to do this (it's a uPy extension), but I can't seem to make it work.  Eg: `obj = array('O', [0x20000000])[0]`

pfalcon, 2015-12-23:
array('O') is array of objects (mp_obj_t[] in C).

pfalcon, 2015-12-23:
> I think that it would be useful to have such a function.

Does Python (as in "CPython") has such a function? If no, then usefulness of such function can rather be  challenged. For comparison, I in my git stash has lot of functions and code which I think would be useful to have, but which I don't burden on MicroPython mainline, because I'm not sure everyone would think the same, and unless I can prove its usefulness, it better sit in my stash. That's how I'd like to see other folks do it - keep working on it next half-year, and if something good comes out of it, push it into mainline then. And to clarify, the key phrase here is not "don't push it for half-year", but "keep working".

Of course, it's not the only way to do it, and I pushed a lot of stuff which I didn't work half-year before, but think are useful to share right away, and other folks may have such stuff too (but from this ticket, I can't figure what's the idea behind this, except that it tries to fish inside raw heap, which is something high-level language should *not* do).

dhylands, 2015-12-23:
I don't really care whether CPython has such a function or not, since I
can't run CPython on my pyboard.

What I want such a function for is for examining the heap. A typical
problem (which doesn't normally exist in CPython) is that the heap gets
full. I'd like to be able to look at objects in my heap to try and
understand what they are, so I can then go and fix my code to be better.

pfalcon, 2015-12-23:
So, do you want something like https://docs.python.org/3/library/gc.html#gc.garbage ?

Otherwise, you see, if you have "function for examining the heap", then running it may (with how you pose requirements for it - will) have effect on heap, so you won't be able to "examine" heap, you will destructively modify it instead. If you want to develop a heap debugger for uPy, that's cool, but starting with adding function you ask for doesn't look like like viable approach. That's why my suggestion - please work on it for sometime, prototype it (of course, keep other folks posted to avoid dup work, and maybe to motivate other folks to join), and when it will be able to do something useful, then push it to mainline, otherwise in half year, there will be just dangling functions of unknown purpose. Of course, I may miss something and what you have in mind may be immediately useful, and I just fail to see it.

dpgeorge, 2015-12-23:
This is how to construct an object given an address (example address is for pyboard start of heap):
```
import array
array.array('O', b'\x40\x31\x00\x20')
```

pfalcon, 2015-12-23:
And general issue is that we miss opportunity to do things in sync and amplify our efforts. Last thing we touched this area (GC, fragmentation), one of contributors wrote a script which showed a heap graphically. It was cool. He didn't submit that script as patch (e.g. for tools/), I wanted to push it nonetheless. Have I done that? Unlikely. So again, I'm not against adding that stuff, I'm for making sure that it's really done up to some useful milestone, and other folks know about it/how to use it (instead of reinvent the wheel next time).

dhylands, 2015-12-23:
> This is how to construct an object given an address (example address is for pyboard start of heap):
> 
> import array
> array.array('O', b'\x40\x31\x00\x20')[0]

That doesn't quite seem to work.
```
>>> foo = bytearray(100)
>>> foo[0:3] = b'abc'
>>> import uctypes
>>> import array
>>> hex(uctypes.addressof(foo))
'0x20006b50'
>>> x = foo
>>> hex(uctypes.addressof(x))
'0x20006b50'
>>> y = array.array('O', '\x50\x6b\x00\x20')[0]
>>> hex(uctypes.addressof(y))
'0x20006f2a'
```

dhylands, 2015-12-23:
> So, do you want something like https://docs.python.org/3/library/gc.html#gc.garbage ?

No.

I want a function (which I called SomeFunction in my hypothetical repl session below), which given an address returns the python object at that address.
```
>>> foo = bytearray(100)
>>> foo[0:3] = b'abc'
>>> import uctypes
>>> import array
>>> hex(uctypes.addressof(foo))
'0x20006b50'
>>> y = SomeFunction(uctypes.addressof(foo))
>>> hex(uctypes.addressof(y))
'0x20006b50'
```

dpgeorge, 2015-12-23:
Probably you need to use bytes, no str, as the second argument to array.array.  You don't need uctypes to get the address.  Eg:
```
>>> foo = bytearray(100)
>>> bytes(array.array('O', [foo]))
b'\xf02\x00 '
>>> array.array('O', b'\xf02\x00 ')[0] is foo
True
```
```

pfalcon, 2015-12-23:
I figure that, I don't see how such function is useful. If such function existed, it would belong to "micropython" module. After writing it, I would wait at least half-year to see if I still find it useful before submitting it upstream.

For comparison: I have a function which returns memoryview into function's value stack. I wrote it a year ago to experiment with JIT compiling and didn't submit it still, no plan to, because w/o complete JIT compiler it's useless (and complete JIT compiler probably wouldn't use it - it was quick hack to get me going with experiments). YMMV.

dhylands, 2015-12-23:
> Probably you need to use bytes, no str, as the second argument to array.array. You don't need uctypes to get the address. Eg:

ok - I typed it too many times and dropped the leading b that time. That seems to work well enough to get started. It has a side effect of allocating memory to do it, but I still think its good enough for experimentation.

I also figured out that using array.array('O') technique returns the real object, where uctypes.addressof returns the address of the buffer, not the address of the object.

Thanks

dhylands, 2015-12-23:
@pfalcon - In the end, I think it will be useful. What my "big" goal to do is to create a heap browser, so I can look at objects in the heap. But I need to start much much smaller. @dpgeorge has given me enough to get started. Once its viable, then I can add a more efficient helper function.

pfalcon, 2015-12-23:
>  where uctypes.addressof returns the address of the buffer, not the address of the object.

Right. id() returns values which directly correlates with address of object, yet still encoded in a way to maximize chance of the value being a small int.

That all still applies to the currently default object representation, if a particular port switches to another representation, any of the above may become invalid.

dhylands, 2016-01-09:
The array.array('O' technique works well - thanks.

Closing this issue.

closed 2016-01-09 by dhylands