micropython/micropython #1748: Prepare for v1.5.2

dpgeorge, 2015-12-24:
Time for the next release!  We had 171 commits since v1.5.1.  Things on the agenda before release:

- [ ] would be nice to merge stmhal CDC improvements, #1713

hoihu, 2015-12-25:
Regarding #1713: I'm happy to help with that.

There are some open points that have not yet been clarified (as pointed out in the pull request). 

I made a measurement with reduced "bInterval" settings in the usb descriptor and it didn't make a difference. At least for the testcases I used. So I believe we can leave that as-is.

Maybe we could add the  testscript (as discussed in #1713) to the unit tests?. It could measure transfer speeds / errors and then fail if it falls below a certain rate.

I did measure the CPU overhead due to the SOF Irq's and it was <0.03%. So even though we have now 1msec irqs it only eats processing time when USB is connected and then also very little. So the "combined approach" I proposed in the pull request is IMO not necessary and would introduce more complexity.

The last thing is the compiler switch and how to best integrate the code changes in master. Any advice?

pfalcon, 2015-12-25:
I usually raise these tickets for new release, but I've kind on purpose didn't do it this time. One reason is that there were many changes lately, and with holidays ahead, I'm not sure how soon/well that can be tested. Another reason is that among the motivations for making a new release was to announce it and find more users/contributors. Well, lately we got influx of bugreports/contributions, and with personal circumstances, I actually found it hard to keep up with them, and IMHO it may be a good idea to at least try to triage them instead of testing next interim release like 1.5.2. Instead, we can go for a bigger release, like 1.6, after holidays. Just my 2 cents.

dpgeorge, 2015-12-26:
Main reason I want a release is because the lite pyboard does not yet have a proper release (ie there is no tagged release with board files for pyblite).  And #1713 is kind of important for that, because it frees up a precious timer for pyblite.

We can save 1.6 for when persistent bytecode is more usable, ie when there exists a cross compiler and a way of doing frozen persistent bytecode (both of these exist in draft form, just need cleaning and pushing).

pfalcon, 2015-12-27:
Sounds good!

dpgeorge, 2015-12-30:
Since #1713 is not ready for merging I'll just go ahead and do a release without it.

dpgeorge, 2015-12-31:
Done!  See ed584e2ffd906e97ae234cccebbc63222a7dbcf2.

closed 2015-12-31 by dpgeorge

pfalcon, 2016-01-03:
1.5.2 docs need to be pushed on http://docs.micropython.org/

dpgeorge, 2016-01-03:
> 1.5.2 docs need to be pushed

I assume you just did it?

pfalcon, 2016-01-03:
Well, I'm not sure how it's organized. Is http://docs.micropython.org/ just alias/proxy for readthedocs site?

dpgeorge, 2016-01-03:
> Is http://docs.micropython.org/ just alias/proxy for readthedocs site?

Yes.  It's still in the to-do queue to switch docs.micropython.org to point to micropython.org/resources/docs (which is auto-built), but blocker for that is writing html/css/js to get a menu working to select ports and versions, and also provide prebuilt docs in pdf form.