micropython/micropython #1752: lib/libm/math function don't heed MICROPY_PY_MATH setting

dhylands, 2015-12-24:
Even with MICROPY_PY_MATH set to zero, all of the the files in lib/libm are still included in the stmhal build.

I'm trying to see if I can get the image down to 128K flash for an STM32F070 processor.

There are a number of solutions:

1 - Create a separate port so that a custom Makefile can be used
2 - Make MICROPY_PY_MATH be a setting in mpconfigboard.mk so that the stmhal Makefile can deal with it.
3 - Add #if's to all of the source files in lib/libm/*.c

I've added options to turn off USB and FATFS (which I currently put in mpconfigboard.mk) and I'm down to 200K I found a bunch of little things that need fixing that I'll do as separate PRs.

I'm planning on doing the USB and FATFS disabling as separate PRs as well.

dpgeorge, 2015-12-24:
With gc sections it shouldn't matter that unused functions are compiled, they won't be linked into the final binary.

dhylands, 2015-12-24:
You're right. I was just being thrown off because they were still showing up in the map file, but they're in a section called "Discarded input sections"

closed 2015-12-24 by dhylands