micropython/micropython #1760: RAM usage for instances with small number of attributes seems excessive

dhylands, 2016-01-02:
I was investigating some memory usage for some classes I was writing, and I was shocked to discover that the instance of a class with a single attribute (which is say a small int) was 80 bytes.

I discovered the following

0 members - 1 heap block - 16 bytes
1-7 members - 5 heap blocks - 80 bytes
8-19 members - 11 heap blocks - 176 bytes

which ties in with the doubling primes table here:
https://github.com/micropython/micropython/blob/master/py/map.c#L50

Classes with more than 19 attributes would seem to be rare (and pylint warns about classes with more than 7 instance variables by default).

I'd like to propose that we add 3 & 5 to the table so that memory consumption of an instance degrades a bit more gracefully when you add attributes to a class. If we did that then we would see the following memory consumption per instance:

0 members - 1 heap block - 16 bytes
1-3 members - 3 heap blocks - 48 bytes
4-5 members - 4 heap blocks - 64 bytes
6-7 members - 5 heap blocks - 80 bytes

I think 2 could also go in the table, but decided not to push my luck.

I understand not wanting to constantly reallocate the table. Maybe this is something where we could have some type of tunable parameter which would allow switching from the doubling table to a table which includes all of the primes up to some number (like 100) and then starts doubling?

If you're writing code which is performance sensitive, then using the doubling table seems reasonable. If you're writing code which is not performance sensitive, but perhaps has lots of instances, then being able to manage memory usage (by having a table with many more smaller entries) seems reasonable.

dpgeorge, 2016-01-03:
It's always going to be a tradeoff.

I agree that 5 heap blocks for a small instance is a lot.  I also think that doubling primes is too aggressive for a hash table allocation scheme for uPy and it should be changed to grow at a smaller rate.

But I don't know if adding small numbers like 3 & 5 to the hash allocation sizes is the best/right way to fix the problem of excessive RAM use for instances.

One optimal (?) way would be to do like they do when JIT'ing: have a variable per type which counts the number of members created at runtime, and then pre-initialise the hash table to have exactly this size.  The first time an instance is created you don't know how many, but subsequent times you do.

Probably we don't want to go to that complexity.  But maybe instead we can delay the initialisation of the members hash table until it's accessed for the first time, and then initialise it to size 2 (1 heap block)?

Or, maybe make the first few "doubling prime" sizes exactly fit in an integral number of blocks, eg `{0, 2, 6, 10, 19, ...}`.  With such small size of hash table it doesn't really matter if it's prime or not.

dhylands, 2016-01-04:
optimal (from a space standpoint) would be to only include as many heap blocks as are needed.
optimal (from a performance standpoint) would be to be able to lookup an entry within a single lookup.

I would imagine that most classes would initialize all of their members in their constructor, but that's only a convention, not a requirement.

One possibility would be to trim the size of the members after the object is constructed. Members could still be added later. But that still doesn't necessarily give you the best performance or best memory usage. And it doesn't cover other dictionary usage.

I still like the idea of having a performance/memory tunable, especially for things which "auto expand".

I was just really shocked to find out that an instance with 8 small-int members took 176 bytes (it doesn't feel very micro). Even optimally, it would still take 64 bytes (4 bytes for each key and 4 bytes for each value). I think that the logical thing for me to do is to separate the code from the data, use the classes for code and use something else for the data (uctypes, struct, one of the array classes or something along those lines). Not as convenient as using members, but gives me more control over memory usage.

It looks like lists double as they expand as well, which is also bad (from a memory usage point of view) for certain list lengths.

Using properties which index into a list would use less memory (although the hit for the first member is still severe).

pfalcon, 2016-04-11:
This is serious issue which affects low-heap ports like microbit or esp8266.

dpgeorge, 2016-04-15:
To address this issue I tried changing the "doubling primes" table to the following: 
```
STATIC uint16_t doubling_primes[] = { 
    0, 2, 4, 6, 8, 10, 12, // +2
    17, 23, 29, 37, 47, 59, 73, // *1.25
    97, 127, 167, 223, 293, 389, 521, 691, 919, 1223, 1627, 2161, // *1.33
    3229, 4831, 7243, 10861, 16273, 24407, 36607, 54907, // *1.5
};
```
This is much less aggressive and the first 7 entries are chosen so they allocate an exact multiple of the GC block (being 4 words).

With this table the size of a user instance is:

0 members - 1 heap block - 16 bytes
1-2 members - 2 heap blocks - 32 bytes
3-4 members - 3 heap blocks - 48 bytes
5-6 members - 4 heap blocks - 64 bytes
7-8 members - 5 heap blocks - 80 bytes
9-10 members - 6 heap blocks - 96 bytes
11-12 members - 7 heap blocks - 112 bytes
13-17 members - 10 heap blocks - 160 bytes

And so on.

Performance wise, pystone on unix does not change (within timing error).

But pystone on pyboard has a 5% decrease, from a score of 1896 down to 1817 (almost zero error on pyboard timing).  That's a large decrease and is because pystone allocates lots of instances of a class (Record) that has 5 members.

If I remove the "4" entry from the doubling_primes table (so it goes `0, 2, 6, 8`) then pystone on pyboard gives 1893, which is much better.

But I don't want to chose the values in the table based just on pystone.  It needs more thinking/investigation.

pfalcon, 2016-04-15:
> But I don't want to chose the values in the table based just on pystone. It needs more thinking/investigation.

Suggestion: rename the table, push 0, 2, 4, 6, 8 change for now.

If you want to go further, making beginning of the table a port define makes sense.

dpgeorge, 2016-04-15:
Above change pushed in 00137b8c11eb94bbf97d5276d9c4e879ea75fb1d (with 0,2,4,6,8,10 list).

Can revisit performance at a later date.

closed 2016-04-15 by dpgeorge