micropython/micropython #1761: uctypes: segfaults with an array of structs?

dhylands, 2016-01-03:
I ran into this just before I went to bed, so I haven't had a chance to look into it yet. I created the following simplfied file:
```python
import uctypes

desc = {
    'f1': (uctypes.ARRAY | 0, 8, {
        'b1': uctypes.UINT8 | 0,
        'b2': uctypes.UINT8 | 1,
    })
}

buf = bytearray(b'0123456789ABCDEF')

S = uctypes.struct(uctypes.addressof(buf), desc, uctypes.LITTLE_ENDIAN)

for i in range(4):
    print('f1[%d].b1 =' % i, hex(S.f1[i].b1));
    print('f1[%d].b2 =' % i, hex(S.f1[i].b2));
```
and it segfaults using unix micropython.

dhylands, 2016-01-03:
Although even with NATIVE, I get a segfault if I use bitfields:
```python
import uctypes

desc = {
    'f2': (uctypes.ARRAY | 4, 8, {
        'b1': uctypes.UINT8 | 0 | 4 << uctypes.BF_POS | 4 << uctypes.BF_LEN,
        'b2': uctypes.UINT8 | 0 | 0 << uctypes.BF_POS | 4 << uctypes.BF_LEN,
    })
}

buf = bytearray(b'0123456789ABCDEF')

S = uctypes.struct(uctypes.addressof(buf), desc, uctypes.NATIVE)

for i in range(4):
    print('f2[%d].b1 =' % i, hex(S.f2[i].b1));
    print('f2[%d].b2 =' % i, hex(S.f2[i].b2));
```

pfalcon, 2016-01-03:
Well, running it thru debugger should show what's wrong.

pfalcon, 2016-01-03:
Note that https://github.com/micropython/micropython/pull/1754 was merged in the meantime, so should be retested.

dhylands, 2016-01-03:
#1754 had no change in behaviour.
#1762 fixes one of the 2 issues that I was running into.

The second issue was at least partly user error.

I used UINT8 instead of BFUINT8. The segfault was caused by trying use the extremely large offset which goes way beyond the end of the provided buffer.

I don't know if trying to access beyond the end of the buffer should raise an error in uPy or not. If it's supposed to catch it, then this issue can remain open (albeit at a low priority).

If having a segfault due to a typo is condiered fine, then we can close this.

dpgeorge, 2022-03-30:
`uctypes` is an advanced module and you can easily segfault the hardware if you pass in bad values.  I don't think it's reasonable for this module to do sanity checking on all input because there are some invalid cases it can never catch.

closed 2022-03-30 by dpgeorge