micropython/micropython #1765: unable to recover from I2C bus error

PeterKenyon, 2016-01-04:
OK this version (1.5.2) has introduced a breaking change for I2C.
It is impossible to recover from an I2C bus error

1.5.1
from pyb import I2C
i2c=I2C(2,I2C.MASTER)
i2c.mem_write(8,76,112) # force a reset causes i2c error (but not until we next do a read)
i2c.init(I2C.MASTER)  # reset the i2c bus to recover
i2c.mem_read(1,76,112) # works

1.5.2
from pyb import I2C
i2c=I2C(2,I2C.MASTER)
i2c.mem_write(8,76,112) # force a reset causes i2c error
i2c.init(I2C.MASTER)  # reset the i2c bus to recover
i2c.mem_read(1,76,112) # nope doesnt work OSError 116
i2c=I2C(2,I2C.MASTER) # see if this works
i2c.mem_read(1,76,112) # nope doesnt work

see:
http://forum.micropython.org/viewtopic.php?f=2&t=1027

dpgeorge, 2016-01-04:
I can't reproduce.  Here is my test script (modified to use I2C bus 1, and init the accelerometer):
```python
import pyb
pyb.Accel()
i2c=pyb.I2C(1,pyb.I2C.MASTER)
try:
    i2c.mem_write(8,76,112)
except OSError as er:
    print(1, repr(er))
i2c.init(i2c.MASTER)
try:
    i2c.mem_read(1,76,112)
except OSError as er:
    print(2, repr(er))
```
It works the same on v1.5.1, v1.5.2 (and master).  It prints nothing (ie no errors).

PeterKenyon, 2016-01-04:
OK thanks Damien, presumably you are using a SCSC16IS7X device, I will see if I can get more info on this, the same software behaves differently depending on 1.5.1 and 1.5.2, Ill be back...

dpgeorge, 2016-01-04:
I don't know what you mean by SCSC16IS7X device... I'm just using a bare PYBv1.0 board connected to a Linux machine over USB VCP.  Please try to write a test script that can be run on stock hardware and prints "good" or "bad" depending on how the I2C behaves.

PeterKenyon, 2016-01-04:
Now you have me confused, the SCSC16IS7X device is the i2c device with address 76, and a write of 8 to internal register 112 will cause a reset which shows anomalous behaviour, yet your code runs without any errors with no device at that address??????

PeterKenyon, 2016-01-04:
Is this related to this
https://github.com/micropython/micropython/issues/1683

dpgeorge, 2016-01-04:
> Now you have me confused, the SCSC16IS7X device is the i2c device with address 76

Haha!  No, I am using just the on-board accelerometer (MMA7660) which just so happens to also have address 76.  So that's why the calls succeed, and why I changed to I2C bus 1.

> Is this related to this #1683

Yes, it could be.

PeterKenyon, 2016-01-05:
OK In summary I think there are 2 anomalies with the I2C implementation.
1. Some I2C writes ie the SC161S7X reset, cause an error which isn't picked up until the next bus access.
2. On Version 1.5.1 this is recoverable by a reset or renew of the i2c object, but 1.5.2 doesn't seem to recover

dpgeorge, 2016-01-05:
Is it possible to make a test that doesn't need the SC161 chip?  For example, connect the I2C bus to some other pins on the pyboard and then fake an I2C slave on these other pins, but make that fake slave cause a bus error?  Maybe it'll need a separate pyboard for the slave.  @PeterKenyon do you know excatly what it is that causes the bus error, is it simply that the nack/ack bit is not sent by the slave?

PeterKenyon, 2016-01-05:
unfortunately I don't have the resources to know exactly what happens, but I think the problem is reproducible with any i2c error. At a guess I would think that setting the reset bit cause the chip to reset immediately so the i2c bus will be left idle for a few ms, so no ACK or NAK, and then waits for a start condition (or something like that). On other i2c buses (BBB) the reset will cause an I2C error on the reset write. I live in Cambridge so if you need any hw ???

PeterKenyon, 2016-01-11:
OK I have found a way to recreate without any hardware, and this doesn't happen on 1.5.1
    from pyb import I2C
    pyb.Accel()
    i2c=I2C(1,I2C.MASTER) 
    i2c.mem_read(1,76,112)  # works
    i2c.mem_read(1,77,112) # fails OSError 5
    i2c.mem_read(1,76,112)  # fails OSError 116
    i2c.init()
    i2c.mem_read(1,76,112) # STILL FAILS !!!
    i2c=I2C(1,I2C.MASTER)
    i2c.mem_read(1,76,112) # STILL FAILS

CRImier, 2016-01-11:
I think I'm affected by this issue as well. After I get any error on the bus, it's impossible to contact the device I've connected to the PyBoard until I power the board off and on again. Just flashed the new firmware yesterday. Will be waiting for this to be fixed =)

dpgeorge, 2016-01-13:
I think I found the problem with this.  Here is an outline of my debugging session:

1. Confirmed bug was introduced between v1.5.1 and v1.5.2 using above test script.
2. Used git bisect to find that it was the commit that automatically turned off the DMA clocks; b677f03407c1b35fcfc4d6948a34a4281033865a
3. But actually this commit just makes the symptoms worse, it's not the cause.
4. Found that by disabling irq's around mem_read calls everything works (no OSError 116 at all).
5. That means it's something to do with I2C DMA calls.
6. Looked in detail at ST's HAL code for doing I2C DMA calls, and found that it sets up DMA before sending slave addr and mem addr, but if there is an error (in sending the addrs) it doesn't deinit the DMA.
7. Moved the DMA initialisation to after the sending of addrs (since it didn't rely on DMA).
8. With my patch (see below) I2C mem_read now works with DMA.  Above tests works as expected.

The patch is:
```diff
--- a/stmhal/hal/f4/src/stm32f4xx_hal_i2c.c
+++ b/stmhal/hal/f4/src/stm32f4xx_hal_i2c.c
@@ -2128,19 +2128,6 @@ HAL_StatusTypeDef HAL_I2C_Mem_Read_DMA(I2C_HandleTypeDef *hi2c, uint16_t DevAddr
     hi2c->State = HAL_I2C_STATE_MEM_BUSY_RX;
     hi2c->ErrorCode = HAL_I2C_ERROR_NONE;
 
-    hi2c->pBuffPtr = pData;
-    hi2c->XferSize = Size;
-    hi2c->XferCount = Size;
-
-    /* Set the I2C DMA transfert complete callback */
-    hi2c->hdmarx->XferCpltCallback = I2C_DMAMemReceiveCplt;
-
-    /* Set the DMA error callback */
-    hi2c->hdmarx->XferErrorCallback = I2C_DMAError;
-
-    /* Enable the DMA Stream */
-    HAL_DMA_Start_IT(hi2c->hdmarx, (uint32_t)&hi2c->Instance->DR, (uint32_t)pData, Size);
-
     /* Send Slave Address and Memory Address */
     if(I2C_RequestMemoryRead(hi2c, DevAddress, MemAddress, MemAddSize, I2C_TIMEOUT_FLAG) != HAL_OK)
     {
@@ -2158,6 +2145,19 @@ HAL_StatusTypeDef HAL_I2C_Mem_Read_DMA(I2C_HandleTypeDef *hi2c, uint16_t DevAddr
       }
     }
 
+    hi2c->pBuffPtr = pData;
+    hi2c->XferSize = Size;
+    hi2c->XferCount = Size;
+
+    /* Set the I2C DMA transfert complete callback */
+    hi2c->hdmarx->XferCpltCallback = I2C_DMAMemReceiveCplt;
+
+    /* Set the DMA error callback */
+    hi2c->hdmarx->XferErrorCallback = I2C_DMAError;
+
+    /* Enable the DMA Stream */
+    HAL_DMA_Start_IT(hi2c->hdmarx, (uint32_t)&hi2c->Instance->DR, (uint32_t)pData, Size);
+
     if(Size == 1)
     {
       /* Disable Acknowledge */
```

And then a similar patch to the `HAL_I2C_Master_Receive_DMA` function also fixes #1683.

So... I guess we need to patch the ST HAL files... Probably not just the F4 but also F2 and F7 files.  Maybe the latest HAL Cube library fixes this problem.  Will need to check.  Should also report this to ST if it's not fixed in their latest code.

dpgeorge, 2016-01-13:
I looked, and it doesn't seem that this is fixed in the latest version of ST's HAL (v1.10.0 from Nov 2015).  That version still initialises DMA before sending the slave addr, and if sending the addr fails then it doesn't reset/deinit the DMA.

dhylands, 2016-01-13:
Presumably we could work around this without modifying the HAL, but aborting the DMA ourselves if the read returns with an error. We could do something like:

tx_dma.State = HAL_DMA_STATE_RESET;
and when HAL_I2C_Mem_Write_DMA returns with an error, if tx_dma.State is HAL_DMA_STATE_BUSY then that means the read started the DMA but didn't abort it, and we can go ahead and abort it.

dpgeorge, 2016-01-13:
Yes, we can do that.  It might also fix other cases where the DMA needs to be reset (eg if the transfer fails part way through).  On the other hand it means we are fixing a problem that doesn't need to be there in the first place (it's a simpler fix to just init the DMA after sending the slave address, rather than before it).

dhylands, 2016-01-13:
If i2c_wait_dma_finished times out we'll also leave the DMA running.

dhylands, 2016-01-13:
It may make sense to have dma_deinit always call HAL_DMA_Abort, just in case.

It looks like aborting just clears the enable bit for the stream, and waits a short while for it to come back as cleared.

Anton-2, 2016-01-13:
On latest STMF7 cube (v 1.3.0), stm32f7xx_hal_i2c.c is v1.0.3 (13 November 2015).
It does a ```HAL_DMA_Abort``` when there is an error sending the address. Nothing like this in HAL_I2C_Mem_Read_DMA however...

dpgeorge, 2016-01-13:
In the meantime, workaround at the Python level is to wrap I2C calls in disable/enable irq.  Only needed for those calls that can fail, eg:
```python
try:
    pyb.disable_irq()
    i2c.mem_read(...)
except OSError:
    # deal with error
finally:
    pyb.enable_irq()
```

dpgeorge, 2016-01-30:
I tried calling HAL_DMA_Abort but it didn't fix the problem.  I tried calling it only when there was a bus error (including timeout waiting for DMA to finish), and unconditionally in dma_deinit.  Neither fixed the problem, ie the I2C transfer following the failed transfer still failed (but actually, if you called I2C.init to reset the bus then it started working again, so doing the abort did do something, but not enough).

Anyway, I fixed it by modifying the HAL source files as I describe above (moving DMA setup to after sending the I2C address); see ea040a4f9ace1c50abc22ab755415305fd2bda41.

closed 2016-01-30 by dpgeorge

dpgeorge, 2016-11-11:
There were still some cases where the I2C peripheral would not be able to recover fully from a bus error (eg slave not sending correct ack/nack during data-bytes transfer).  I made a few patches to help fix this, see e6da6a720fb2bb48c6885e5f28f5f917690080fd and 74fb5d6932c835f92c6985799e30ea28b6449705.  But still DMA I2C transfers are not fully robust to errors so in c4e58eaa9820f5e63d9d94cec5ffb403c1dffbe7 I added an option to the I2C constructor (and init() method) to allow to allow/disallow DMA on the I2C.  DMA is now *disabled* by default for I2C (which is not really an issue, I2C does not need precise timing like SPI does sometimes).