micropython/micropython #1769: Is there a method to get the binary bytes from an inline asm function?

dhylands, 2016-01-07:
I was recently writing an inline assembler function in python (I finally did get it to work), but I ran into some issues that made it difficult to debug.

My first attempt ran into a hard fault. Even with the hard fault information being printed, that didn't help a whole lot, since the PC is located somewhere in the middle of the heap.

It would be really good if there was some way of getting at 2 pieces of information:
1 - the address of the assembler function
2 - the length of the assembler function

With those 2 pieces of information, I could extract the bytes into a file, transfer said file to the host, and disassemble it.

To get at that particular bit of information, I wound up having to hack emitglue.c and enable DEBUG_PRINT (and point DEBUG_printf to printf) and write an stmhal version of WRITE_CODE:
```C
    FIL fp;
    f_open(&fp, "/flash/out-code", FA_WRITE | FA_CREATE_ALWAYS);
    UINT n;
    f_write(&fp, fun_data, fun_len, &n);
    f_close(&fp);
```
then I could transfer out-code back to the host and I used the following to disassemble it:
 (I was able to use the command: 
```bash
arm-none-eabi-objdump -bbinary -marm -D out-code -Mforce-thumb
```
Now I could finally figure out the offset within the routine given the PC and starting address of the code.

This seems like a rather arduous procedure, so I thought I'd see if there is any desire to have this simplfied? Even if I could just get the address, I could probably figure out an approximate size from the heap, and then use uctypes to grab the actual bytes.

I was also thinking it would useful to capture the Hard Fault info (the register values that it prints) and stash it away in a reserved chunk of RAM so that you can press RESET and recover it without having to hook up a UART.

dpgeorge, 2016-01-07:
Try the following:
```python
import stm
import array

def inspect(f, ninstr=16):
    @micropython.asm_thumb
    def dummy():
        pass
    if type(f) != type(dummy):
        raise ValueError('expecting an inline-assembler function')
    baddr = bytes(array.array('O', [f]))
    addr = baddr[0] | baddr[1] << 8 | baddr[2] << 16 | baddr[3] << 24
    print('function object at: 0x%08x' % addr)
    print('number of args: %u' % stm.mem32[addr + 4])
    code_addr = stm.mem32[addr + 8]
    print('machine code at: 0x%08x' % code_addr)
    print('machine code dump:')
    for i in range(ninstr):
        a = code_addr + 2 * i
        print('  %04x  %04x' % (a, stm.mem16[a]))
```

dpgeorge, 2016-01-07:
Once the assembler code is compiled the length is discarded (it's not needed to execute the function, so why waste keeping it around?) so there is no way to get at this information.  You'll need to just inspect the machine code for as long as you think the function runs.

dhylands, 2016-01-07:
Cool - that works for me. I put together a PR #1770 which basically just wraps the gc_nbytes function which will give an approximation of the size.

I'm planning on a writing a forum article titles something along the lines of "Debugging Hard Faults in inline assembler" and give some tips.

dhylands, 2016-01-09:
I posted my article in the forum. 
http://forum.micropython.org/viewtopic.php?f=2&t=1295

Thanks very much for the python snippet.
Closing this issue.

closed 2016-01-09 by dhylands