micropython/micropython #1771: verbose output order

pohmelie, 2016-01-07:
When I try to run unix build `./micropython -v -v test.py` to see bytecode:
```
$ ./micropython -v -v test.py 
File test.py, code block '<module>' (descriptor: 0x7fc62524df40, bytecode @0x7fc62524e2e0 34 bytes)
Raw bytecode (code_info_size=10, bytecode_size=24):

 05 01 00 00 00 00 0a 82 32 83 7a 24 47 00 00 00 ff 3f 04 00 44 35 08 80 32 32 32 45 35 01 80 41 11 5barg names:(N_STATE 5)
(N_EXC_STACK 1)
  bc=-1 line=1
  bc=3 line=2
  bc=10 line=4

mem: total=2606, current=674, peak=2260
stack: 32 out of 80000
GC: total: 2072832, used: 736, free: 2072096
 No. of 1-blocks: 5, 2-blocks: 2, max blk sz: 6
00 SETUP_EXCEPT 703 POP_BLOCK04 JUMP 1507 POP_TOP08 POP_TOP09 POP_TOP10 POP_EXCEPT11 JUMP 1514 END_FINALLY15 LOAD_CONST_NONE16 RETURN_VALUE
```
There is missed order of strings, which were printed before new line, but appears later. As http://stackoverflow.com/questions/7431425/c-stdout-print-without-new-line said, I insert `fflush(stdout);` between https://github.com/micropython/micropython/blob/master/py/showbc.c#L560-L561 and right after https://github.com/micropython/micropython/blob/master/py/showbc.c#L112. And after output looks like:
```
$ ./micropython -v -v test.py 
File test.py, code block '<module>' (descriptor: 0x7f067086af40, bytecode @0x7f067086b2e0 34 bytes)
Raw bytecode (code_info_size=10, bytecode_size=24):
 05 01 00 00 00 00 0a 82 32 83 7a 24 47 00 00 00
 ff 3f 04 00 44 35 08 80 32 32 32 45 35 01 80 41
 11 5b

arg names:(N_STATE 5)
(N_EXC_STACK 1)
  bc=-1 line=1
  bc=3 line=2
  bc=10 line=4
00 SETUP_EXCEPT 7
03 POP_BLOCK
04 JUMP 15
07 POP_TOP
08 POP_TOP
09 POP_TOP
10 POP_EXCEPT
11 JUMP 15
14 END_FINALLY
15 LOAD_CONST_NONE
16 RETURN_VALUE
mem: total=2606, current=674, peak=2260
stack: 32 out of 80000
GC: total: 2072832, used: 736, free: 2072096
 No. of 1-blocks: 5, 2-blocks: 2, max blk sz: 6
```

Don't really know is this a real issue, or just my system problem. System is xubuntu 14.04.

pohmelie, 2016-01-08:
Same thing here: https://github.com/micropython/micropython/blob/master/py/parse.c#L277-L284

pfalcon, 2016-01-08:
This was a known issue which could as well have fixed itself, but apparently it didn't. The idea is that we should not use libc stdio framework, but instead native uPy output framework. And actually, printf() was redirected to mp_printf() even for unix port, that's why this issue should have fixed itself. If it didn't, need to investigate further.

dpgeorge, 2016-01-29:
Fixed, see PR linked above.

closed 2016-01-29 by dpgeorge