micropython/micropython #1773: How to communicate directly with the rs485 port

xhat, 2016-01-09:
I have a serial sensor with rs485, how to communicate directly with pyboard?

AWCharlie, 2016-01-09:
RS485 is differential signalling, usually over twisted pair. You have three
options:

1: A "smart circuit" that detects direction (Tx or Rx) and appropriate
delays in your communication code. These circuits have a small delay as
they detect a bit and turn on transmit mode.
2: Use a GPIO pin to control transmit or receive.
3: Use a duplex setup with 2 twisted pair.

Everyone makes RS-485 interface chips these days. The basic unit is like
the TI SN76176B half duplex or the Max485. The Max485 is easy to use. You
need your serial Tx and Rx to be logic level and one digital GPIO to
control direction. These chips have  transmit control pin and a receive
control pin that can be tied together and to your GPIO. You have to turn on
transmit before you send characters and turn off after so that received
characters will be buffered.

xhat, 2016-01-09:
thx..
now i have a max13487eesa Module( http://www.alibaba.com/product-detail/SCM-TTL-to-RS485-module-485_60235101990.html ), but still not find the correct way to use the module,

import pyb
ua = pyb.UART(4, 2400, timeout_char=1, read_buf_len=1000)
ua.write('F5F5')
ua.read(10)

nothing reply. can u give a demo code ?

dhylands, 2016-01-09:
Are you meaning to send the two hex bytes 0xf5 0xf5 or are you meaning to send 4 ASCII caharacters 'F', '5', 'F', and '5'. Your code does the latter.

What device is this connected to? Are you using compatable baud rates, stop bits, and bits/character?

Were the above lines typed at the REPL? or part of a program?

pfalcon, 2016-01-09:
Guys, this bugtracker is for MicroPython *issues*. General discussion of MicroPython usage happens at the forum, http://forum.micropython.org/ . Please move the communication on this topic there.

xhat, 2016-01-09:
thx all. i have add a new topic at the forum.

closed 2016-01-09 by dpgeorge