micropython/micropython #1774: socket accept returns a bytearray rather than a tuple

dhylands, 2016-01-10:
When I do:
```
        conn, addr = s.accept()
        print('Connection from:', addr)
        print('type of conn', type(conn))
        print('type of addr', type(addr))
```
under CPython, I get back the following:
```
Connection from: ('127.0.0.1', 57423)
type of conn <class 'socket.socket'>
type of addr <class 'tuple'>
```
When I do the same thing under MicroPython, I get back:
```
Connection from: bytearray(b'\x02\x00\xe0Y\x7f\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00')
type of conn <class 'socket'>
type of addr <class 'bytearray'>
```
I can see the values (57423 = 0xE04F, which is the \xe0Y, and \x7f\x00\x00\x01 is the 127.0.0.1).
So my question is, is this worth fixing?
I can check for bytearray being returned and extract the values myself if need be. I just figured I'd ask first.

pfalcon, 2016-01-11:
You don't provide enough context. See https://github.com/micropython/micropython/issues/1375 . Consequently, the most likely cause is not implemented feature in micropython-lib (random guess based on lack of context).

danicampora, 2016-01-11:
Is that on the unix version or on the WiPy?

dhylands, 2016-01-11:
My entire program was:
```
import socket
IP_ADDR = '127.0.0.1'
IP_PORT = 8888
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
s.bind((IP_ADDR, IP_PORT))
s.listen(1)
conn, addr = s.accept()
print('Connection from:', addr)
print('type of conn', type(conn))
print('type of addr', type(addr))
```
and this was running the unix version of micropython.

pfalcon, 2016-01-11:
Partial fix pushed as https://github.com/micropython/micropython-lib/commit/f1533338eab99e15fc1583b90d904a9c80d9acbb . It says what's required for full fix.

pfalcon, 2016-01-11:
With https://github.com/micropython/micropython/pull/1775 , fully fixed in micropython-lib.

closed 2016-01-11 by pfalcon