micropython/micropython #1778: property does not accept keyword arguments

chrysn, 2016-01-13:
the property function does not accept keyword arguments, making micropython incompatible to software that documents itself like this:

    class A:
        x = property(lambda self: self._x, doc="The X property")

i'm preparing a patch, but am not yet sure whether it is wiser to just silently discard kwargs (after all, they are mostly used for `doc=` in property), to check for the kwargs to be only `doc` and otherwise err out, or to do full kwarg handling.

i'm leaning towards full kwarg handling, as long as it does not bring too big a code increase.

dpgeorge, 2016-01-13:
I assume you ran into this problem in some real world code?

chrysn, 2016-01-13:
yes, though of my own writing. it shows up a lot in other places (judging by [debian codesearch](https://codesearch.debian.net/results/filetype:py%20property.%2Adoc=), and it's not only the standard library, although i couldn't filter that out easily).

dpgeorge, 2016-01-19:
Now implemented by #1780.

closed 2016-01-19 by dpgeorge