micropython/micropython #1787: segfaults after gc_collect

stinos, 2016-01-18:
````
MicroPython v1.4.3-1119-gac16cc9 on 2016-01-18; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import gc
>>> gc.collect()
5
>>> gc.collect()
3
Segmentation fault (core dumped)
````

I'll have to investigate this further to find the actual cause, but from what I can see now (on Windows, where it segfaults in another place) it lookas as if ```mp_state_ctx_t.vm.mp_module_builtins_override_dict``` gets sweeped in the first call to gc_collect(), so on the second execution the lookup in mp_load_global() segfaults because mp_module_builtins_override_dict doesn't contain a valid dict anymore.

dpgeorge, 2016-01-18:
Wow.  I can confirm it's a bug (on Linux) and git-bisect tells me it started with 3ca84026db59166441c5b33af12fe2826793e66b.

dpgeorge, 2016-01-18:
But git-bisect only tells a half truth: compiling that commit with DEBUG=1 leads to no seg fault, so that commit is just the first time it shows up easily.  It must have been introduced before that.

stinos, 2016-01-18:
Might be since 035a0a2

stinos, 2016-01-18:
Likely, but it's hard to see what exactly goes wrong. In any case, forget what I said earlier: it's not the sweeping: we probably both use the up arrow to enter the second gc.collect() call in the REPL, which results in readline_process_char calling vstr_add_str(rl.line, MP_STATE_PORT(readline_hist)[rl.hist_cur]); and *there* it overwrites mp_module_builtins_override_dict with garbage..

Anton-2, 2016-01-18:
I don't know if it helps, but the results on OS X are slightly different : 
```code
MicroPython v1.5.2-68-gac16cc9 on 2016-01-18; darwin version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import gc
>>> gc.collect()
6
>>> gc.collect()
3
>>> gc.collect()
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
>>> <press Ctrl-d to exit>
Assertion failed: (!"bad free"), function gc_free, file ../py/gc.c, line 478.
Abort trap: 6
``` 

The results are the same if I type everything, or use the history (up arrow) to recall gc.collect().
After a simple assignment (```a=0```) before the failing gc.collect(), I can call gc.collect() many times with no error. But there is still a bad free on exit.

dpgeorge, 2016-01-18:
Ok, found the bug: the GC is not scanning the entire stack.

Symptom (seen in this issue): the readline vstr in unix/main.c/do_repl() contains a pointer to its buffer.  This pointer lives on the stack.  The GC is not scanning this part of the stack and so its reclaiming the vstr buffer.  The memory is then reused for mp_module_builtins_override_dict.

Cause: mp_stack_ctrl_init() sets the top of the stack used for GC scanning.  But it's being called too deep and setting the stack_top pointer way below the actual top.  The reason for this is that unix/main.c/main() is inlining the do_repl() function (it's static) and so the readline vstr in do_repl() actually lives on the stack frame within main().

This could be endemic across all ports.  We need to set the stack top to truly the stack top.

stinos, 2016-01-18:
> setting the stack_top pointer way below the actual top

Makes sense. The reason I used the code above was to debug the gc because I was getting segfaults in gc_collect due to ```((uintptr_t)MP_STATE_VM(stack_top) - (uintptr_t)&regs) / sizeof(uintptr_t)``` being unreasonably large at some point, but I couldn't pinpoint where that suddenly came from.

pfalcon, 2016-01-18:
> We need to set the stack top to truly the stack top.

Should be reverted then: 8a96ebea75158987c1136f1a618d98cf387445fd

dpgeorge, 2016-01-18:
> Should be reverted then: 8a96ebe

Doesn't make a difference.  As I said above, the pointer in question lives within the same stack frame as which calls stack_ctrl_init.  Even the following won't work:
```c
static int main2() {
    // do usual stuff here
}

int main() {
    mp_stack_ctrl_init();
    main2();
}
```
It won't work because main2 is static.  Making it non static might help but we will still depend on the compiler (and linker) not optimising it.

pfalcon, 2016-01-18:
> Should be reverted then:

But even then there's no guarantee that top will be captured right. Should have main ... dup

pfalcon, 2016-01-18:
> Making it non static might help but we will still depend on the compiler (and linker) not optimising it.

It must not optimize it, unless aggressive compile-time options are set (whole-program optimization).

stinos, 2016-01-18:
> Doesn't make a difference

well it might not change the big picture, but moving mp_stack_ctrl_init to the first line in main does fix this particular problem I'm having (and the code shown as well)

pfalcon, 2016-01-29:
Fixed by https://github.com/micropython/micropython/pull/1805.

closed 2016-01-29 by pfalcon