micropython/micropython #1789: Implement stm32f439 Hardware crypto

david-jimenez, 2016-01-20:
@ryannathans has been working on a port for the 439 based on the 429 disco and I'll be working on the crypto wrapper for it.

How should the interface work? Is there any existing documentation for micropython? And would we like the 439 port config files posted up too?

dpgeorge, 2016-01-20:
> How should the interface work?

On the Python side it should be exposed through a standard, existing Python library.  If possible...

> Is there any existing documentation for micropython?

Use the code in stmhal/ as reference.

> And would we like the 439 port config files posted up too?

If they are as simple as adding a directory in stmhal/boards/, then yes please!

ryannathans, 2016-01-20:
@dpgeorge will do a push and PR tomorrow for the /board files when I'm not on mobile.

The port isn't based on a publicly available board like the DISCO, we're using a "bare" chip for its numerous UART/SPI, encryption and cost. For us in bulk, the 439 is cheaper than the 405.

dpgeorge, 2017-06-14:
Closing due to inactivity.

closed 2017-06-14 by dpgeorge