micropython/micropython #1791: STMHAL from the cube to the py

rolandvs, 2016-01-21:
Dear all,

Is it to be expected that the current STM HAL version for the F4xx (v1.1.0) (19-jun-2014) will be updated towards the latest from STM (1.4.1) (9-oct-2015)?

Is somebody already working on this?

It seems that STM did an overhaul of the HAL code and it would be nice to keep up...

dpgeorge, 2016-01-26:
@rolandvs It's quite a bit of effort to update the HAL, and creates lots of git "noise", so best to only do it when necessary.  Are there significant changes/improvements in the latest version?

rolandvs, 2016-01-26:
@dpgeorge There have been a number of updates in the Cube libraries of STM. It would be nice to be able to keep up with that. One thing is that they have renamed a lot and added a legacy .h file to stay compatible. I did not check for significant changes (yet) and I realize that it indeed generates a lot gitish noise.When is it necessary? Well there is mention of a revamped structure for hardware #1430. So that would be a good point in time...

At this point I cannot promise anything, but I do want to make an effort to look at the STM HAL. But first I want to be more proficient with microPython in the weeks to come. ;-)

dpgeorge, 2016-11-22:
The HAL has been updated for the F4 MCUs to v1.5.2.  CMSIS has also been updated.

closed 2016-11-22 by dpgeorge

dpgeorge, 2017-08-29:
The STM HAL is now maintained in the repository at https://github.com/micropython/stm32lib which should make it easier to update in the future.