micropython/micropython #1793: STM32F439 "bare-bones" Port

ryannathans, 2016-01-22:
I've made considerable changes to the STM32F429-DISCO to support the STM32F439, which is identical to the 429 except it has hardware accelerated hashing/crypto.

As the DISCO is significantly different to the "bare" 429/439 I think these changes should eventually be merged.

My modifications so far compared to STM32F429-DISCO:
* Changed linker to support correct memory ranges (DISCO has 8MB SDRAM)
* Added CCM to linker (missing in DISCO)
* Increased heap size (STM32F4x9 has 64k more ram than STM32F405, not sure why this wasn't done in the DISCO)
* Increased core clock to 192MHz (12MHz overclock to support USB, instead of 12MHz underclock seen in DISCO)
* Added ADC123_IN0 to PA0, something that seems to be missing in the DISCO AF file.

david-jimenez, 2016-01-24:
@ryannathans Looks like the linker file that you modified this from has been changed from the default one. Your declaration of FLASH_TEXT doesn't match that of the comment accompanying it, you're grabbing an additional 32K of the next sector alongside it

ryannathans, 2016-02-03:
@david-jimenez will fix, must have been a typo in the DISCO linker

ryannathans, 2016-02-25:
#1857

closed 2016-03-28 by ryannathans