micropython/micropython #1796: Repeated ctrl-c sequence hard crashes pyboard

hoihu, 2016-01-22:
came up during development of #1713 but seems to be an independant issue.

The following let the pyboard hard-crash after app. 100 tries on the current master.
```python
#!/usr/bin/env python

import time
import serial

write_test_script = """
import pyb
while (1):
    pyb.LED(3).toggle()
"""
        
def drain_input(ser):
    time.sleep(0.1)
    while ser.inWaiting() > 0:
        print('drain:', ser.read(ser.inWaiting())[:20])
            
port = """/dev/cu.usbmodem1452"""
    
def main():
    dev = serial.Serial(port, baudrate=4000000)
    print("starting")
    for i in range(0,10000):
        print("try {}".format(i))
        dev.write(b'\x03\x03\x01\x04') # break, break, raw-repl, soft-reboot
        drain_input(dev)
        dev.write(bytes(write_test_script.encode('ascii')))
        dev.write(b'\x04') # eof
        dev.flush()        
    
if __name__ == "__main__":
    main() 
```
@dhylands proposed in #1713 that it could be caused if 

> ..if an interrupt to process Control-C happens between gc_init and pyb_usb_dev_init0 the interrupt
> handler will be referencing uninitialized data (most likely previously initialized data). However, since that
>  early initialization always happens in the same order it seems likely that the exception object will get
> allocated at the same location each time (having the object get trampled by something else is usually
> what causes the serious issues).

I'm not sure if a simple flag can be used to signal if a reset/startup is pending. So the ctrl-c sequence is only executed if the pyboard has finished booting up.

dpgeorge, 2016-01-26:
Thanks @hoihu!  I can confirm that the above script leads to a hard crash, and it only takes 10-50 tries.

hoihu, 2016-01-26:
Do you have an idea of how to fix it? I believe it's not a critical issue but it blocks #1713.

dpgeorge, 2016-01-26:
I'm looking at it now.  It is related to double ctrl-c, and is hence related to PENDSV ISR.  I've disable the PENDSV stuff and it works fine.  Using the hard-fault debug code provided by @dhylands (it's very useful!) it looks like the fault is due to jumping to a location without having the Thumb-code bit set (ie LSB in the PC).  It's strange.

dpgeorge, 2016-01-26:
I can't make progress on this, but the problem goes away with the following hack:
```diff
--- a/stmhal/pendsv.c
+++ b/stmhal/pendsv.c
@@ -56,9 +56,11 @@ void pendsv_nlr_jump(void *o) {
     if (MP_STATE_VM(mp_pending_exception) == MP_OBJ_NULL) {
         MP_STATE_VM(mp_pending_exception) = o;
     } else {
+        /*
         MP_STATE_VM(mp_pending_exception) = MP_OBJ_NULL;
         pendsv_object = o;
         SCB->ICSR = SCB_ICSR_PENDSVSET_Msk;
+        */
     }
 }
 
```
Ie disable PENDSV support.  @hoihu can you test #1713 with this patch to see if the problems disappear?  If so then we can merge #1713 and fix this particular bug later.

hoihu, 2016-01-27:
Ok thanks, will try tonight.

What I also want to check is if the hard crash occurs if CTRL-D is not sent after double CTRL-C's.

Btw what could happen if ctrld is sent first then followed up by ctrlc immediately afterwards? repl is exit in main.c followed by the soft reset, but Tim3 remains active - so ctrlc is still evaluated although the VM is in the init process. Could that lead to a potential issue as well?

hoihu, 2016-09-29:
This is an old bug and perhaps no longer important.

Perhaps one improvement to fix "CTRL-c during soft reboot" may be to switch off irqs that handle incoming characters (e.g. the SOF USB irq or Uart Rx Irq) just after the VM repl terminates. And re-enable them just before the VM starts. 

In fact any enabled peripheral Irq may be disabled during that time since the (python-)irq handler are still in place and may produce undesired actions?

@dpgeorge do you consider to pursuit this or should we close it?

dpgeorge, 2016-09-30:
No, we should leave this open, it's still a bug that needs fixing.

iabdalkader, 2019-09-18:
This is most likely related to the LDM/STM issue #3842

jonnor, 2024-09-29:
Last comment was 5 years go. Is this issue still present?

robert-hh, 2024-09-29:
A quick test with a current firmware shows that it is not present any more. So let's close it. Thank @jonnor for digging through the old issues. Most of them should be obsolete by now.

closed 2024-09-29 by robert-hh