micropython/micropython #1801: sdcard driver assertion failure

peterhinch, 2016-01-28:
This problem may be in the calling code as the sdcard driver looks fine to me. But it only occurs when writing to an SD card device rather than to the SD card on the Pyboard. File foo.py:
```python
import sdcard, pyb
s = str(''.join(('a' for _ in range(1024)))) # fails at >= 1024
sd = sdcard.SDCard(pyb.SPI(1),pyb.Pin.board.X21)
try:
    pyb.mount(sd, '/fc')
except OSError:
    print('Probably already mounted')
fn = '/fc/test'
with open(fn, 'w') as f:
    f.write(s)
try:
    pyb.mount(None, '/fc')
except OSError:
    print('Failed to umount')
```
Outcome:
```python
>>> import foo
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "foo.py", line 16, in <module>
  File "foo.py", line 16, in <module>
  File "sdcard.py", line 196, in writeblocks
AssertionError: 
>>> 
```
I have checked that the SD card contains a valid filesystem and that strings of < 1024 bytes are written and can be read back. The assertion failure appears to be a block length > 512 bytes which is puzzling as I've used the block protocol on various devices without encountering this issue.

dpgeorge, 2016-01-28:
There is one place in fatfs/ff.c that can write more than 1 block, and that's in f_write when you pass it a buffer larger than the block size.  So that's definitely a possibility here.

A few things:

1. The traceback in foo.py seems to come from the end of the script... I would have thought it failed in the f.write call.  Can you check this?
2. Did you try the same code writing to the pyboard's built in SD card?  (I think that's what you said at the start of your post).

peterhinch, 2016-01-28:
tl;dr
1. It is coming from f.write - see below
2. It does work with the SD card - again see below
I've used the block protocol with FRAM devices, with my own Flash driver and with a pseudo device based on a file; in the case of the Flash device frequently storing files of a variety of sizes. This is the first time I've seen this error. I'm baffled as to why it only occurs with the sdcard device. Here's a session that hopefully demonstrates the issue.
```python
>>> s = str(''.join(('a' for _ in range(1024)))) # fails at >= 1024
>>> sd = sdcard.SDCard(pyb.SPI(1),pyb.Pin.board.X21)
>>> pyb.mount(sd, '/fc')
>>> import os
>>> os.listdir('/fc')
['tr45', 'main.py', 'flash.py', 'fontmap.py', 'myfont.py', 'rats.py', 'rats1.py', 'pete', 'myfont', 'rats2.py', 'ea_2_7.py', 'image.bin', 'epaper.py', 'epd.py', 'ea_2_7.xbm', 'cat_2_7.xbm', 'text_image_2_7.xbm', 'rats3.py', 'smallfont', 'inconsolata16x10', 'Courier_NewTilde_is_degs19x11', '507295820', 'timesroman45x46', 'test.bin', 'block.py', 'blocks', 'data', 'inconsolata', 'boot.py', 'schedsupport.py', 'ea_2_7.bup', 'usched.py', 'schedtest.py', 'LiberationSerif-Regular45x44', 'test.py', 'panel.py', 'catface.xbm', 'fram.py', 'micropower.py', 'rm.py', 'lo.py', 'bmp180.py', 'nrf24l01.py', 'upower.py', 'radio.py', 'structures.py', 'pickle.py', 'message.py', 'arial19x19', 'arial13x13', 'arial16x16', 'Courier_New19x11$isDegs', 'Arial14ptBold19x19$isdeg', 'Arial19x19$isdeg', 'rats', '507296386', 'test']
>>> fn = '/fc/test'
>>> f = open(fn, 'w')
>>> f.write(s)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "sdcard.py", line 196, in writeblocks
AssertionError: 
>>> 
>>> f.close()
>>> fn = '/sd/test'
>>> f = open(fn, 'w')
>>> f.write(s)
1024
>>> f.close()
>>> f = open(fn, 'r')
>>> f.read()
'aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa'
>>> f.close()
>>> 
```

dpgeorge, 2016-01-29:
Ok, the problem is simply that f_write can indeed make a call to the underlying device to write multiple blocks at once.

I put a printf into stmhal/diskio.c/disk_write() and indeed I see that the number of sectors written is 2 or more when a buffer of 1024 or more bytes is written to a file (eg 4 sectors with 2048 bytes).  I also verified that reads can read more than one block at once (eg calling file.read(2048) does indeed read 4 blocks at once, but only if the file itself uses contiguous sectors).

Summary: block drivers should support multiple sector read/writes.

So, how shall we support multiple blocks in sdcard.py?  There are 2 options:

1. Call single block read/write in a loop n times.  This is easy to implement.
2. Implement multiple block read/write.  This is harder to implement but gives better performance.  Is it guaranteed that all SD cards support multiple block transfers?  I assume so.

It would be nice to implement option 2.  This page has a good description of how to do multiple block transfers: http://elm-chan.org/docs/mmc/mmc_e.html

peterhinch, 2016-01-29:
OK, this explains the results I've had - the drivers I cited support multiple block r/w, and my successful use of sdcard has been for small random access reads from large files. I'm guessing such reads only use single block transfers.

I'll make an attempt at option 2.

dpgeorge, 2016-02-04:
Fixed by 2bd758fe96e2913f48c1a0fdccafddc3dfb31cf9.

closed 2016-02-04 by dpgeorge