micropython/micropython #1803: Prepare for v1.6

dpgeorge, 2016-01-28:
It's time again for the next release, which will be v1.6.  It should be released this weekend.  Things on the list:

- [ ] addition of mpytool.py to inspect .mpy files
- [ ] ability to use mpytool.py to freeze .mpy files so precompiled Python code can be executed from flash (basically this is "true" frozen modules, although I think it's good to keep the existing "frozen script" frozen modules as well)

pfalcon, 2016-01-28:
> although I think it's good to keep the existing "frozen script" frozen modules as well

I hope yes, until .mpy use will become widespread ;-).

dpgeorge, 2016-01-28:
> although I think it's good to keep the existing "frozen script" frozen modules as well
> 
> I hope yes, until .mpy use will become widespread ;-).

Also because it's (currently) a complicated issue to get a uPy cross compiler built for the target.  It's nice to be able to freeze raw text files: it just works.

dpgeorge, 2016-01-31:
mpy files and persistent bytecode is too big a change and will need to wait for next version (but is available at #1811 for those wanting to try it out).

dpgeorge, 2016-01-31:
v1.6 has been tagged here: 331a48195d325b9557175056e2c3650ef496d862

closed 2016-01-31 by dpgeorge