micropython/micropython #1806: Merge code from "MicroPython on ESP8266" Kickstarter campaign

dpgeorge, 2016-01-30:
@pfalcon and I are currently running a Kickstarter campaign to fund development of the ESP8266 port.  Hopefully the campaign is successful!  In that case we will have a lot of code to merge, and along with that will come a lot of bug fixes and features for ESP8266.  For info see:

https://www.kickstarter.com/projects/214379695/micropython-on-the-esp8266-beautifully-easy-iot

This issue exists to discuss general things about merging the code (eg when it'll be done) and to reference from other ESP8266 issues.

AWCharlie, 2016-01-30:
Read the KS details yesterday. The proposal looks good. Might do the
corporate sponsorship level (whatever it is called :-)  )  ESP32 uses the
same RISC IIRC. Target that at the same time? (Got a guy on the inside
there).

dpgeorge, 2016-01-30:
@AWCharlie any and all contributions are much appreciated, thank you!  We do talk about the ESP32 on the KS page.  It's something we are interested in -- and it will benefit from this work -- but want to make the ESP8266 rock solid first.

roger-, 2016-01-30:
There's an [active ESP branch](https://github.com/mianos/micropython) with a bunch of fix/improvements -- any thoughts of merging that too?

dpgeorge, 2016-01-30:
@roger- Yes, there's a lot of good code there, but it remains to see how easy it will be to merge into our new execution model and framework.

roger-, 2016-01-30:
@dpgeorge: just saw on KS that you guys are planning to go with traditional sockets. Have you thought about instead integrating a scheduler with `async`/`await` support (a la [curio](https://curio.readthedocs.org/en/latest/))? 

Being able to do `data = await socket.recv(1000)` natively would be awesome, particularly if interrupts and peripherals worked that way too.

dpgeorge, 2016-01-30:
> just saw on KS that you guys are planning to go with traditional sockets.

Before traditional sockets you have to first think of how standard Python work, just a for loop or something like that.  Python's execution model assumes exclusive control over the task that it runs in, and that's at odds with the ESP8266's native execution model of callbacks.  If we didn't first fix the execution model then somewhere down the track people's scripts would just not work if they did too much processing before "yielding" (however the yield works, whether it be an await on a socket, a sleep_ms, or a complete return from execution a la JavaScript).

Once we have a "standard" execution model then we automatically get standard sockets.  From there we can possible implement await/async/yield/coroutines, but the matter would always remain: to allow an arbitrary Python scripts to run you can assume they will eventually yield, and so can't have the underlying system be callback/event based.

roger-, 2016-01-30:
> Python's execution model assumes exclusive control over the task that it runs in, and that's at odds with the ESP8266's

Ah right, I get that. I just thought taking an exclusively async approach from the start (while still axing the old  callback model) might be more efficient than having to do  sync and later async.

Anyway it's an new exciting project, and `async` support would be a great selling point if you're considering another stretch goal  :)

pfalcon, 2016-01-30:
@roger- : There's also thread on a forum which you may find interesting, and it links to HackerNews discussion which I find particularly insightful (as usual).

dpgeorge, 2016-01-30:
> I just thought taking an exclusively async approach from the start (while still axing the old callback model) might be more efficient than having to do sync and later async.

This is an interesting idea which I personally did not consider.

pfalcon, 2016-01-30:
@AWCharlie :

> ESP32 uses the same RISC IIRC. Target that at the same time? (Got a guy on the inside there).

Why, that could be nice 50K stretch goal ;-). Seriously speaking, how can we pledge to support chip which probably only couple of dozens of people in the world played with so far? Couple of hundreds should get it, but I don't know if everyone got so far, and there's still road from getting to playing with it. I got mine, but it's a DIY kit, and requires some fine SMT soldering. I don't have my rework station with me now, and feel reserved to do soldering with a simple small-power iron, for if something goes wrong, I won't be able to fix it or just break. And then it's still ESP31, not ESP32. And we already know that final ESP32 will have differences in pinout, and if they went for such change, just imagine how much will change on the insides. Summing up, ESP32 would be quite different project with different timeframe and type of effort required. We went for ESP8266 project to avoid that "moving target" syndrome, and based on experience already collected for ESP8266 for more than a year, with chip at peoples' disposal, and with timeframe and scope which are both realistic and should be interesting for most people. It will enable future ESP32 work in the same way that work done over the last year enables the current project, but we can't give any specific figures or pledges at this time (hardly anyone can). If that makes sense, we'll sure appreciate your support.

Neon22, 2016-01-30:
Thanks so much for doing this guys. I think this will expand the audience considerably.
I just wanted to say Thanks for all your continued effrots. Backed it.

pfalcon, 2016-01-30:
> This is an interesting idea which I personally did not consider.

We want to start from the start, from normal sockets which everyone knows (and which any beginner should learn first). @roger- : You're well aware that MicroPython has uasyncio. I hope you're aware that @dpgeorge implemented initial support for `async` keyword, which @pohmelie now extends, so it'll be merged when passes review cycle. So, it'll be avaiable for esp8266 in a natural course of action, as long as the port is maintained. What we want to achieve with the KS project is to deliver solid, useful base, within useful timeframe (few months), while securing further support of the platform for a useful time again (order of a year). The more successful we're with the campaign, the farther we can go beyond the basics (direction(s) to go will be steered by the community, and on distant roads, we'll need community help with development, testing for sure).

roger-, 2016-01-30:
>  What we want to achieve with the KS project is to deliver solid, useful base, within useful timeframe

Absolutely, just thought native, integrated async/coros might be more performant and appropriate for consideration at this early stage (beginners would just have to drop in a `await` prefix) :)

Congrats on the KS success BTW!

mianos, 2016-02-02:
Even if you don't want the API my fork uses it would probably be very easy to grab the low level bit banged I2C and 1wire HALs. I have a HAL and python interface in seperate files.
The few super ESP specific things like the wifi  smartconfig module would be worth keeping on the esp module as no such functionality exists in the wipy or pboards and it is super handy.
(Personally I have abandoned the socket interface in modest.c and written a complete web server and client abstraction, like the rest of the python world. I just pulled master a few days ago and hopefully I will try and follow the ESP changes).

pfalcon, 2016-05-28:
@mianos : Every contribution to MicroPython is welcome. And every contribution should follow a process which is standard for most open-source projects: preparing a pull request, undergoing a review, adjusting based on the feedback, rinsing and repeating. Thanks!

pfalcon, 2016-05-28:
The code was merged a while ago, closing.

closed 2016-05-28 by pfalcon