micropython/micropython #1807: pyb.mount() feature suggestion

peterhinch, 2016-01-30:
When testing code using block protocol devices an issue arises where a coding error causes the program to fail between mounting the device and umounting it. In this instance the device remains mounted. The program then fails on a subsequent run with an ``OSError`` "device already mounted". While this can be circumvented by trapping the ``OSError``, it's not ideal.

It would be better if the ``mount`` command implemented the context manager protocol.

dpgeorge, 2016-02-04:
> It would be better if the mount command implemented the context manager protocol.

You mean thot mount() should call `__exit__()` upon failure?

peterhinch, 2016-02-04:
Sorry, I hadn't explained this well. The context manager protocol means that __exit__() is called under all circumstances; I envisaged __exit__() unmounting the device, so that one could write
```python
with mount(device, mountpoint):
   with open('rats', 'r') as f:
      # use the file
```
At this point the file would have been closed and the device then unmounted, regardless of whether an error had occurred or not.

pfalcon, 2017-05-13:
pyb.mount() is a legacy function, maintained as is. There's new VFS subsystem in MicroPython now (optimized for code size, as usual).

closed 2017-05-13 by pfalcon