micropython/micropython #1809: docs: pyboard online docs missing some content

peterhinch, 2016-01-30:
If I build the docs locally the outcome is correct. However the online docs omit some recently added detail for example the ``ADCAll`` documentation in library/pyb.ADC.rst [here](http://docs.micropython.org/en/latest/library/pyb.ADC.html) and the line regarding the firmware build in library/network.rst [here](http://docs.micropython.org/en/latest/library/network.html)

pfalcon, 2016-01-30:
@peterhinch : ReadTheDocs doesn't build docs automatically. @dpgeorge build new version after release. But you should be able to trigger build of "latest" docs yourself (I can). On http://docs.micropython.org/ , lick ReadTheDocs at bottom right, in sliding menu select "Build", should be clear from there. I rebuild docs, and see both additions (shift+reload in browser if in doubt!)

Thanks for docs contributions!

peterhinch, 2016-01-30:
I appreciate there's a delay before online docs are updated. In this instance there has been a regression. The online docs were updated - now some (not all) of the changes are lost. Work has been done in the interim to better support multiple boards and I suspect an error here has resulted in this knock-on effect.

pfalcon, 2016-01-30:
@peterhinch : I'm not sure what you mean, can you elaborate which changes are lost and what knock-on effect do you mean?

peterhinch, 2016-01-31:
This problem has either been fixed, or has gone away: the pages now display correctly here. Possibly some local internet issue?

closed 2016-01-31 by peterhinch