micropython/micropython #1812: Assertion failed when trying to access non-exsiting attributes

stinos, 2016-02-01:
Instead of using hasattr I somewhere ended up with code similar to

```
for objName, objId in locals().items():
  try:
    myAttr = objId.SomeAttribute
    # use myAttr 
  except:
    continue
````

This results in

```micropython: ../py/vm.c:1071: mp_execute_bytecode: Assertion `exc_sp == exc_stack - 1' failed.
Aborted (core dumped)```

pfalcon, 2016-02-01:
I'd suspect that continue in except, all wrapped in for may be an issue here. Can you try to separate that from locals() and AttributeError?

stinos, 2016-02-01:
Indeed changing ```continue``` to ```pass``` gets rid of the assert. Using a 'normal' dictionary instead of locals doesn't change anything, nor does catching AttributeError. So for reference, this works as expected:
```
someDict = { 'a': 0 }
for _, objId in someDict.items():
  try:
    myAttr = objId.someAttr
  except AttributeError:
    pass
```

but this has the assertion failing:
```
someDict = { 'a': 0 }
for _, objId in someDict.items():
  try:
    myAttr = objId.someAttr
  except AttributeError:
    continue
```

dpgeorge, 2016-02-01:
Fixed, see above.  @stinos please verify and close this issue :)

stinos, 2016-02-01:
Verified, thanks!

closed 2016-02-01 by stinos