micropython/micropython #1813: mpz: Bug in bignum division routine for x86-64

dpgeorge, 2016-02-01:
Unix build of uPy for x86-64 has some bugs in the bignum division routine, which can be seen by running the following code (which should print 6 lines with 1 on each line):
```python
x = 0x8000000000000000
print((x + 1) // x)
print((x + 1) % x)
x = 0x86c60128feff5330
print((x + 1) // x)
print((x + 1) % x)
x = 111618804455139159771299797594024125819032083630825107882289957652821831815073
print((x + 1) // x)
print((x + 1) % x)
```
It's something to do with shifting (at the C level) a 32-bit number by 32 bits, in mpn_div, but needs investigation.

alex-march, 2016-02-10:
Interestingly, running two almost the same tests, produces different results:
```python
x = 0x8000000000000000
# print(hex((x + 1) // x))
print(hex((x + 1) % x))
>>> 0x100000001
```
And if I uncomment the first print, the output is:
```python
>>> 0x180000001
```
Having a quick look, mpn_div seems to be doing it's job up until we start casting mpz_dig_t into mpz_dbl_dig_t in [mpz.c#L568](https://github.com/micropython/micropython/blob/master/py/mpz.c#L568) and shifting.

dpgeorge, 2016-05-08:
Fixed by dc3faea0405dea803828f5a2be314734b8c166b6.

closed 2016-05-08 by dpgeorge