micropython/micropython #1814: Frozen modules don't support packages

pfalcon, 2016-02-01:
E.g., it's not possible to import (module not found) urllib.urequest if it's frozen.

dmazzella, 2016-05-08:
news about this?

dpgeorge, 2016-05-13:
@dmazzella no news.  Do you have a need for this feature?

pfalcon, 2016-05-14:
7b1bf0c3087aa87cde7417b95b4dd3f8e4ca3781 alleviates problem a bit.

dmazzella, 2016-05-15:
@dpgeorge yes i need it, i have a complx library wrote for cPy and with same patch (io.BytesIO.tell, hasattr and other) i  have  successfully port it to uPy. But it only work for Unix port becouse the size off the library don't fit pyboard ram. For do it i need forzen modules to work with submodules for don't rewrite a complex library that i already have.

dpgeorge, 2016-05-16:
> For do it i need forzen modules to work with submodules

Are you using frozen bytecode (scripts compiled to .mpy using mpy-cross), or just frozen scripts (the raw script text is stored in flash with the firmware)?

If the latter then @pfalcon's latest patch may work for you.  But I suspect you are using the former (frozen bytecode) because that helps you reduce RAM usage.

dmazzella, 2016-05-16:
I would use the .mpy files (frozen bytecode)

dpgeorge, 2016-05-23:
Frozen packages should now be fully supported, with both frozen scripts and frozen mpy files.

@dmazzella can you confirm that it now works for you?  Just put your entire directory structure of py files in stmhal/frozen/ (or whatever you call it) and recompile.

ryannathans, 2016-05-24:
This commit (3e03d1b87e81f0960067519e0713f70eb6e9927c) breaks frozen compiling for me. I have all my .py files in a folder called scripts. I am compiling with the following command

`make -j 12 BOARD=STM32F439 FROZEN_MPY_DIR=scripts -C stmhal`

this worked fine until this commit, now I get the following error:

```
CC usbdev/class/src/usbd_msc_scsi.c
CC usbdev/class/src/usbd_msc_data.c
CC build-STM32F439/pins_STM32F439.c
Creating build-STM32F439/frozen_mpy.c
usage: mpy-tool.py [-h] [-d] [-f] [-q QSTR_HEADER]
                   [-mlongint-impl {none,longlong,mpz}] [-mmpz-dig-size N]
                   files [files ...]
mpy-tool.py: error: the following arguments are required: files
Makefile:301: recipe for target 'build-STM32F439/frozen_mpy.c' failed
make: *** [build-STM32F439/frozen_mpy.c] Error 2
make: *** Deleting file 'build-STM32F439/frozen_mpy.c'
make: *** Waiting for unfinished jobs....
```

ryannathans, 2016-05-24:
Furthermore, reverting that one individual commit results in all modules having a parent of the folder they're stored in for compiling. For example, all my modules now have to be imported by "import scripts.foo" instead of "import foo". This behavior might have been generated by me reverting the Makefile so I could get past compilation though...

dpgeorge, 2016-05-24:
Please rebuild mpy-cross.  And then run make with V=1 to see what it's doing.

ryannathans, 2016-05-25:
```
CC usbdev/class/src/usbd_msc_data.c
arm-none-eabi-gcc -I../lib/netutils -I. -I.. -Ibuild-STM32F439 -Icmsis/inc -Icms
is/devinc -Ihal/f4/inc -Iusbdev/core/inc -Iusbdev/class/inc -I../lib/mp-readline
 -I../lib/netutils -I../lib/timeutils -Wall -Wpointer-arith -Werror -ansi -std=g
nu99 -nostdlib  -DSTM32F439xx -mthumb -mabi=aapcs-linux -mfpu=fpv4-sp-d16 -mfloa
t-abi=hard -fsingle-precision-constant -Wdouble-promotion -mtune=cortex-m4 -mcpu
=cortex-m4 -DMCU_SERIES_F4 -Os -DNDEBUG -Iboards/STM32F439 -DSTM32_HAL_H='<stm32
f4xx_hal.h>' -fdata-sections -ffunction-sections -DMICROPY_QSTR_EXTRA_POOL=mp_qs
tr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -c -MD -o build-STM32F439/usbde
v/class/src/usbd_msc_data.o usbdev/class/src/usbd_msc_data.c
CC build-STM32F439/pins_STM32F439.c
arm-none-eabi-gcc -I../lib/netutils -I. -I.. -Ibuild-STM32F439 -Icmsis/inc -Icms
is/devinc -Ihal/f4/inc -Iusbdev/core/inc -Iusbdev/class/inc -I../lib/mp-readline
 -I../lib/netutils -I../lib/timeutils -Wall -Wpointer-arith -Werror -ansi -std=g
nu99 -nostdlib  -DSTM32F439xx -mthumb -mabi=aapcs-linux -mfpu=fpv4-sp-d16 -mfloa
t-abi=hard -fsingle-precision-constant -Wdouble-promotion -mtune=cortex-m4 -mcpu
=cortex-m4 -DMCU_SERIES_F4 -Os -DNDEBUG -Iboards/STM32F439 -DSTM32_HAL_H='<stm32
f4xx_hal.h>' -fdata-sections -ffunction-sections -DMICROPY_QSTR_EXTRA_POOL=mp_qs
tr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -c -MD -o build-STM32F439/pins_
STM32F439.o build-STM32F439/pins_STM32F439.c
Creating build-STM32F439/frozen_mpy.c
python ../tools/mpy-tool.py -f -q build-STM32F439/genhdr/qstrdefs.preprocessed.h
  > build-STM32F439/frozen_mpy.c
usage: mpy-tool.py [-h] [-d] [-f] [-q QSTR_HEADER]
                   [-mlongint-impl {none,longlong,mpz}] [-mmpz-dig-size N]
                   files [files ...]
mpy-tool.py: error: the following arguments are required: files
Makefile:301: recipe for target 'build-STM32F439/frozen_mpy.c' failed
make: *** [build-STM32F439/frozen_mpy.c] Error 2
make: *** Deleting file 'build-STM32F439/frozen_mpy.c'
make: *** Waiting for unfinished jobs....
```

ryannathans, 2016-05-25:
In older versions before that commit it looks like this
```

STM32F439.o build-STM32F439/pins_STM32F439.c
Creating build-STM32F439/scripts/frozen_mpy.c
python ../tools/mpy-tool.py -f -q build-STM32F439/genhdr/qstrdefs.preprocessed.h
 build-STM32F439/scripts/plcboot.mpy build-STM32F439/scripts/function.mpy build-
STM32F439/scripts/gda_manager.mpy build-STM32F439/scripts/canbus.mpy build-STM32
F439/scripts/gps.mpy build-STM32F439/scripts/datamanager.mpy build-STM32F439/scr
ipts/basetask.mpy build-STM32F439/scripts/temp.mpy build-STM32F439/scripts/pins.
mpy build-STM32F439/scripts/system.mpy build-STM32F439/scripts/scheduler.mpy bui
ld-STM32F439/scripts/tpms.mpy build-STM32F439/scripts/sharedmem.mpy build-STM32F
439/scripts/telit.mpy build-STM32F439/scripts/ds18b20.mpy build-STM32F439/script
s/onewire.mpy build-STM32F439/scripts/master.mpy build-STM32F439/scripts/hexdeco
.mpy build-STM32F439/scripts/plc.mpy build-STM32F439/scripts/wdt.mpy build-STM32
F439/scripts/obm.mpy build-STM32F439/scripts/router.mpy build-STM32F439/scripts/
serial.mpy build-STM32F439/scripts/netapps.mpy build-STM32F439/scripts/ut.mpy bu
ild-STM32F439/scripts/slave.mpy > build-STM32F439/scripts/frozen_mpy.c
CC build-STM32F439/scripts/frozen_mpy.c
arm-none-eabi-gcc -I../lib/netutils -I. -I.. -Ibuild-STM32F439 -Icmsis/inc -Icms
is/devinc -Ihal/f4/inc -Iusbdev/core/inc -Iusbdev/class/inc -I../lib/mp-readline
 -I../lib/netutils -I../lib/timeutils -Wall -Wpointer-arith -Werror -ansi -std=g
nu99 -nostdlib  -DSTM32F439xx -mthumb -mabi=aapcs-linux -mfpu=fpv4-sp-d16 -mfloa
t-abi=hard -fsingle-precision-constant -Wdouble-promotion -mtune=cortex-m4 -mcpu
=cortex-m4 -DMCU_SERIES_F4 -Os -DNDEBUG -Iboards/STM32F439 -DSTM32_HAL_H='<stm32
f4xx_hal.h>' -fdata-sections -ffunction-sections -DMICROPY_QSTR_EXTRA_POOL=mp_qs
tr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -c -MD -o build-STM32F439/scrip
ts/frozen_mpy.o build-STM32F439/scripts/frozen_mpy.c
LINK build-STM32F439/firmware.elf
```

dpgeorge, 2016-05-25:
It works for me.  What is your build system?  What does `make -p BOARD=STM32F439 FROZEN_MPY_DIR=scripts|grep FROZEN` give?  What does `find scripts/ -type f -name '*.py'` give?  (executed in stmhal directory)

ryannathans, 2016-05-26:
@dpgeorge I'm running Windows/cygwin. 

When I run with the commit 3e03d1b included I get the following error
```
D:\WTGit\EmbeddedMicroPython\stmhal>make -p BOARD=STM32F439 FROZEN_MPY_DIR=scripts|grep FROZEN
FIND: Parameter format not correct
```

When I run without that commit I get the following

```
D:\WTGit\EmbeddedMicroPython\stmhal>make -p BOARD=STM32F439 FROZEN_MPY_DIR=scripts|grep FROZEN
FROZEN_MPY_DIR = scripts
OBJ =  $(PY_O) $(addprefix $(BUILD)/, $(SRC_LIB:.c=.o)) $(addprefix $(BUILD)/, $
(SRC_C:.c=.o)) $(addprefix $(BUILD)/, $(SRC_O)) $(addprefix $(BUILD)/, $(SRC_HAL
:.c=.o)) $(addprefix $(BUILD)/, $(SRC_USBDEV:.c=.o)) $(addprefix $(BUILD)/, $(SR
C_MOD:.c=.o)) $(BUILD)/pins_$(BOARD).o $(BUILD)/$(FROZEN_MPY_DIR)/frozen_mpy.o
CFLAGS = $(INC) -Wall -Wpointer-arith -Werror -ansi -std=gnu99 -nostdlib $(CFLAG
S_MOD) -D$(CMSIS_MCU) $(CFLAGS_MCU_$(MCU_SERIES)) $(COPT) -Iboards/$(BOARD) -DST
M32_HAL_H='<stm32$(MCU_SERIES)xx_hal.h>' -fdata-sections -ffunction-sections -DM
ICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY
-*-command-variables-*- := FROZEN_MPY_DIR=scripts BOARD=STM32F439
FROZEN_MPY_MPY_FILES := build-STM32F439/scripts/plcboot.mpy build-STM32F439/scri
pts/function.mpy build-STM32F439/scripts/gda_manager.mpy build-STM32F439/scripts
/canbus.mpy build-STM32F439/scripts/gps.mpy build-STM32F439/scripts/datamanager.
mpy build-STM32F439/scripts/basetask.mpy build-STM32F439/scripts/temp.mpy build-
STM32F439/scripts/pins.mpy build-STM32F439/scripts/system.mpy build-STM32F439/sc
ripts/scheduler.mpy build-STM32F439/scripts/tpms.mpy build-STM32F439/scripts/sha
redmem.mpy build-STM32F439/scripts/telit.mpy build-STM32F439/scripts/ds18b20.mpy
 build-STM32F439/scripts/onewire.mpy build-STM32F439/scripts/master.mpy build-ST
M32F439/scripts/hexdeco.mpy build-STM32F439/scripts/plc.mpy build-STM32F439/scri
pts/wdt.mpy build-STM32F439/scripts/obm.mpy build-STM32F439/scripts/router.mpy b
uild-STM32F439/scripts/serial.mpy build-STM32F439/scripts/netapps.mpy build-STM3
2F439/scripts/ut.mpy build-STM32F439/scripts/slave.mpy
FROZEN_MPY_PY_FILES := scripts/plcboot.py scripts/function.py scripts/gda_manage
r.py scripts/canbus.py scripts/gps.py scripts/datamanager.py scripts/basetask.py
 scripts/temp.py scripts/pins.py scripts/system.py scripts/scheduler.py scripts/
tpms.py scripts/sharedmem.py scripts/telit.py scripts/ds18b20.py scripts/onewire
.py scripts/master.py scripts/hexdeco.py scripts/plc.py scripts/wdt.py scripts/o
bm.py scripts/router.py scripts/serial.py scripts/netapps.py scripts/ut.py scrip
ts/slave.py
        $(Q)$(PYTHON) $(MPY_TOOL) -f -q $(BUILD)/genhdr/qstrdefs.preprocessed.h
$(FROZEN_MPY_MPY_FILES) > $@

D:\WTGit\EmbeddedMicroPython\stmhal>
```

Running the other command gives the same message on both
```
D:\WTGit\EmbeddedMicroPython\stmhal>find scripts/ -type f -name '*.py'
File not found - '*.py'
```

however

```
D:\WTGit\EmbeddedMicroPython\stmhal>dir scripts
 Volume in drive D is New Volume

 Directory of D:\WTGit\EmbeddedMicroPython\stmhal\scripts

25/05/2016  12:41 PM    <DIR>          .
25/05/2016  12:41 PM    <DIR>          ..
25/05/2016  12:41 PM               863 basetask.py
25/05/2016  12:41 PM             2,139 canbus.py
25/05/2016  12:41 PM             1,712 datamanager.py
25/05/2016  12:41 PM             4,077 ds18b20.py
25/05/2016  12:41 PM             1,072 function.py
25/05/2016  12:41 PM             1,505 gda_manager.py
25/05/2016  12:41 PM            13,237 gps.py
25/05/2016  12:41 PM             9,522 hexdeco.py
25/05/2016  12:41 PM            14,114 master.py
25/05/2016  12:41 PM            10,375 netapps.py
25/05/2016  12:41 PM             2,655 obm.py
25/05/2016  12:41 PM             8,181 onewire.py
25/05/2016  12:41 PM               405 pins.py
25/05/2016  12:41 PM             1,733 plc.py
25/05/2016  12:41 PM            23,837 plcboot.py
25/05/2016  12:41 PM             8,868 router.py
25/05/2016  12:41 PM             4,689 scheduler.py
25/05/2016  12:41 PM               657 serial.py
25/05/2016  12:41 PM             1,366 sharedmem.py
25/05/2016  12:41 PM             1,987 slave.py
25/05/2016  12:41 PM             5,668 system.py
25/05/2016  12:41 PM               525 telit.py
25/05/2016  12:41 PM             2,160 temp.py
25/05/2016  12:41 PM             9,779 tpms.py
25/05/2016  12:41 PM             1,502 ut.py
25/05/2016  12:41 PM             2,230 wdt.py
              26 File(s)        134,858 bytes
               2 Dir(s)  3,874,828,210,176 bytes free

D:\WTGit\EmbeddedMicroPython\stmhal>
```

dpgeorge, 2016-05-26:
@ryannathans please try executing the following: `find scripts/ -type f -name "*.py"` (note double quotes).  If it works you can change line 287 of stmhal/Makefile to use double quotes instead of single.

stinos, 2016-05-26:
> D:\WTGit\EmbeddedMicroPython\stmhal>find scripts/ -type f -name '*.py'
File not found - '*.py'

that's not the cygwin shell, but the windows command processor and seeing the output it uses windows' find which is very different from *nix find.

dpgeorge, 2016-05-26:
> that's not the cygwin shell, but the windows command processor and seeing the output it might be using windows' find which is very different from *nix find.

Ok.  Any suggestions on a portable way to do a recursive wildcard match in a makefile?

stinos, 2016-05-26:
Using a cygwin or msys shell is the way to go I think, since all common *nix-like utilities are available there and use the same conventions.

But actually I do not know why ryannathans is having problems, the find command works fine for me even when on wincows cmd:
```
d:\some\dir\micropython> C:\cygwin\bin\find.exe -type f -name '*.py'
./cc3200/boards/make-pins.py
./cc3200/tools/smoke.py
...
```

pfalcon, 2016-05-26:
@dpgeorge: @stinos hints that the ryannathans runs wrong things in wrong place.

ryannathans, 2016-05-26:
Changing line 287 of the Makefile to the following results in a successful build

`FROZEN_MPY_PY_FILES := $(shell C:\cygwin64\bin\find.exe $(FROZEN_MPY_DIR)/ -type f -name '*.py')`

dpgeorge, 2016-05-28:
> Changing line 287 of the Makefile to the following results in a successful build

Ok, so we don't change anything in the Makefile.

I guess then that frozen modules (both frozen scripts and frozen bytecode) now support packages.

closed 2016-05-28 by dpgeorge