micropython/micropython #1817: RFC: frame buffer module to support fast display drivers

dpgeorge, 2016-02-04:
Pyboards (PYB, WiPy, ESP8266, etc) should be able to drive displays like LCDs and OLEDs.  It's currently possible to write such a driver in pure Python, and even get decent performance for small screens.  But large displays (even 320x240) have a lot of pixels and doing even simple operations like filling a rectange can take a lot of time written in pure Python.

So, on the one hand, one wants display drivers written in C.

On the other hand, we want to make it easy for people to write a driver to support any display, and hence should keep writing in Python.

One idea to get the best of both worlds is to have a frame buffer module (framebuf?) which does the intensive memory copy operations, and then leave the actual interfacing to a small Python driver (eg I2C, SPI interface).  This is similar in philosophy to the filesystem and the block driver protocol.

framebuf would support different colour formats (eg black and white of different bit layouts, greyscale, and RGB) so that it stores data in a way that is native to the display.  It would support primitive drawing operations, region copying and font blitting.

The underlying hardware driver just needs to init the display and send the frame buffer data out on the wire.

I already implemented a proof-of-concept of this idea for the ESP8266, using an SSD1306 OLED display, and it works well.

Comments welcome!  In particular, @peterhinch I'm interested if/how this could help with your epaper driver.

danicampora, 2016-02-04:
Very good idea Damien!
How much does this frame buffer module add to the code size?

dpgeorge, 2016-02-04:
> How much does this frame buffer module add to the code size?

At the moment not much, but that's because it only supports a single colour format and has only a few functions.  It's only a proof-of-concept and needs a lot of design and coding to make it good for general purpose use.

peterhinch, 2016-02-04:
Regarding e-paper drivers these are binary devices so the mapping is one bit per pixel. In the case of the unit I'm using it's mapped such that the MSB of byte 0 is at (x,y) (0,0) with LSB at (7,0). Byte 1 maps to (8,0) - (15,0). Obviously other mappings are possible: making it generic could be chllenging. If the framebuffer class can handle this sort of arrangement it would indeed be useful for e-paper displays.

Clearly they are atypical devices in this respect. Another aspect, at least of the one I'm using, is that display update is slow: delays are specified in the device datasheet. So there is less need for high speed display handing. On the other hand, surely faster devices will become available.

In general, fast support for display devices would be a superb feature.

dpgeorge, 2016-02-04:
> In the case of the unit I'm using it's mapped such that the MSB of byte 0 is at (x,y) (0,0) with LSB at (7,0).

That makes sense, I'd call it horizontal mapping.  The LCD skin for the pyboard uses vertical mapping with LSB at the top (ie LSB of first byte is (0,0), MSB of first byte is (0, 7)).  The SSD1306 also uses vertical mapping.  Probably horizontal and vertical would cover most black & white displays.

> Another aspect, at least of the one I'm using, is that display update is slow: delays are specified in the device datasheet. So there is less need for high speed display handing.

The idea is that the frame buffer is stored in RAM on the MCU and operations like text rendering and rectangles, lines, circles etc are done quickly using the frame buffer C module.  But the writing of the frame buffer to the device uses Python.  So your epaper display would benefit because it would have much faster buffer drawing code.

Note that this entire scheme only works if the frame buffer can fit in RAM.  This is true for small screens up to about 320x240 for black & white (320*240/8 = 9600 bytes).

If the frame buffer doesn't fit in RAM then a totally different scheme is needed.  You'd have to generate the frame buffer on the fly as you write it to the device (eg read an image from the SD card), or the device will need to support rendering operations like set pixel etc.

Anton-2, 2016-02-04:
I'm using the 480x272 24 bit LCD screen of the STM32F7 DISCO with a driver done fully in micropython, with very satisfying results. But it's a slightly different case : the frame buffer is in external SDRAM, and I'm using the "Chrom-ART" accelerator for bit blits, rectangle fill, color conversion and blending.

I this situation, I would be interested in a C 'frame buffer' library providing : 

- Vector draw primitives. Currently I'm drawing line in python, with a tiny Bresenham inside loop in assembly. But having already written (and C fast) curves, wide lines and maybe beziers would be very valuable.

- "Pixmap" loading and converting. I'm using an had hoc format to load pictures elements from SDCard (mostly a frame buffer dump with some headers). Having a common format (with eventually some RLE) would be handy.

Of course, this library should provide bit blit, blending and color conversion when there is no hardware support...

peterhinch, 2016-02-05:
@Damien Your OP discusses bus connected devices. Is it worth also considering memory mapped displays as can be found on targets other than the Pyboard? Small displays are usually bus connected whereas larger ones can be memory mapped for reasons of bandwidth. In the former case matching the FB layout to that of the display is a matter of efficiency rather than logical necessity. In the second case it's mandatory with the memory address, pixel and colour mapping being hardware defined. Ideally a FB class would be capable of coping with both but perhaps this is a generalisation too far.

dhylands, 2016-02-05:
I used a framework (about 15 years ago) called MicroWindows (now called Nano-X). It allows the drivers to separated from the drawing code and has a fairly nice interface for plugging in new drivers.

It also has support for touch screens, mice, and keyboards, and seems to have framebuffer drivers which support linux and Win32 (so could also be used from say the unix version of MicroPython).

Nano-X can be found here: http://www.microwindows.org

I've used it with displays which have their own framebuffers (i.e. the small display) and with displays which use shared memory (coming from SDRAM).

So I thought I would mention it. You can use the pieces you want and don't necessarily have to use the entire framework.

peterhinch, 2016-02-23:
An issue relevant to this discussion is the storage of fonts. The normal approach of importing a bytes object containing the entire font is fast but costly in RAM. In my e-paper display I use a random access file and pull in single characters - cheap in RAM but slow.

It would be good if fonts could be stored in, and accessed from, flash. A possible way might be to use C. But it could be good if there be a way of storing an array in flash (like frozen modules). If there were a way of declaring a reference to a bytes object located in flash then elements could be accessed directly without needing RAM.

pfalcon, 2016-02-23:
To my shame, I still didn't have a chance to read this RFC. But I had few cases where quick comments I was too shy to record were simply forgotten, so I'd rather dump my mind. So, to develop truly generic framebuffer interface, we should consider issue of hardware acceleration and multi-buffering (e.g. single-buffering vs double-buffering). To support double+-buffering, there already should be .swap() call. On single-buffer, it would be a nop, but on double-buffer that's what will cause framebuffer changes to be presented to user. There're bunch more questions in that regard of course, e.g. after .swap(), what is the contents of the current buffer (the most generic solution is to say that it's underspecified, but of course, that's not ideal solution for single-buffer).

peterhinch, 2016-02-24:
On the topic of fonts or other data in flash, will the forthcoming persistent byte code fix this? Say I write a Python module that declares a global bytes object, and then make it persistent. If I import the module is the bytes object accessed directly from flash or pulled into RAM?

dpgeorge, 2016-02-24:
> On the topic of fonts or other data in flash, will the forthcoming persistent byte code fix this?

Yes.  So we don't need to provide a separate, special case for "frozen" fonts.

> Say I write a Python module that declares a global bytes object, and then make it persistent. If I import the module is the bytes object accessed directly from flash or pulled into RAM?

It'll stay in flash and be accessed from there.  Note that persistent(=frozen) bytecode (as will be available soon) requires you to rebuild the firmware because the frozen module is baked into the firmware.  To support "dynamic freezing" (where you can freeze a given module from the prompt) is doable but not on the TODO list just yet :)

peterhinch, 2016-02-24:
Excellent! My interest stems from [this thread](http://forum.micropython.org/viewtopic.php?f=14&t=1396) where @Roberthh is doing some great work interfacing TFT displays to the Pyboard. We are discussing ways improve font handing without losing RAM or performance.

dpgeorge, 2016-04-12:
Initially framebuf module was implemented in 53ad681ed1786601ca5eccb60f8fd950b8bc47d7.

peterhinch, 2016-04-13:
I'll look at that. Out of interest have you seen [this](https://github.com/robert-hh/SSD1963-TFT-Library-for-PyBoard.git) colour touchscreen driver from @robert-hh? The hardware supports its own graphics RAM and uses a parallel interface to the Pyboard which he's programmed in assembler. It supports screens up to 800x480. The performance is excellent, the main drawback is that it uses a lot of pins. RAM usage is low.

peterhinch, 2016-04-13:
Re framebuf. A potential enhancement is text rendering: alas on high DPI displays an 8x8 font is barely legible. Ideally it would support fonts of arbitrary dimensions, and offer both fixed and proportionally spaced rendering. The e-paper driver and that of @robert-hh provide a route for converting system fonts and ttf files to formats matched to the respective driver giving users flexibility.

Another factor to consider is maintaining a text insertion pointer to handle newline characters and (optionally) line wrapping. This frees the user from having to calculate the coordinates for entering text which can be nontrivial with proportional fonts. So the driver has two methods, one to set the coordinates of the insertion pointer and another to render text in a given font. Consecutive calls to the second method populate the buffer with text, respecting newlines.

There are many ways this could be taken further, but this level of functionality is usable while still being reasonably "micro".

Incidentally the route we use for converting fonts involves a closed source (but free as in beer) program GLCD Font Creator. This outputs C source which our utilities convert to Python or binary files. An open source equivalent would be preferable if anyone knows of one.

mcauser, 2016-08-29:
My 2c. The petme font `font_petme128_8x8.h` is slighly too wide to fit "temperature" on my Nokia 5110 (84x48).
Hoping we can switch to / add a less-bold 5x7 font, such as this Adafruit one:
https://github.com/adafruit/Adafruit-GFX-Library/blob/master/glcdfont.c

robert-hh, 2016-08-29:
Hello Mike. The library mentioned above by Peter (https://github.com/robert-hh/SSD1963-TFT-Library-for-PyBoard) also includes some tools to take & convert fonts from your computer. It Makes use if a GLCD font converter of MicroElectronica, which creates C files, and then we use a python srcipt to convert that to a proper python structure. The smallest font I've created was 6x8. The font files in my package are bit packed row-by-row, wheresas the output of the GLCD converter is column-by-column non-packed, as far as I recall. So you may have a look at  that approach for other fonts.

peterhinch, 2016-08-30:
@robert-hh Some while ago you pointed out an approach to font creation using open source tools: [freetype](http://freetype-py.readthedocs.io/en/latest/index.html). This is on my list but a solution would require some study.

kamikaze, 2016-10-14:
I'm going to try writing an SSD1322 driver. How can I use framebuf?

mcauser, 2016-10-15:
This tool can help you with the fonts. It support both rowMajor to columnMajor.
http://www.eran.io/the-dot-factory-an-lcd-font-and-image-generator/

peterhinch, 2016-10-15:
A couple of points. It's no longer actively maintained and, though it claims to run under Linux it does seem Windows orientated in that, as far as I can see from the docs, the fonts it converts are Windows fonts.

I'd rather a solution which works with open font formats. My reading suggests that [Freetype](http://freetype-py.readthedocs.io/en/latest/introduction.html) is the most widely used open source solution, it has Python bindings and it is installed by default in most Linux systems. See this brief [wikipedia article](https://en.wikipedia.org/wiki/FreeType). Writing a generic interface to framebuffers has been on my todo list for some time.

mkarliner, 2016-10-15:
That would be a wonderful thing.

peterhinch, 2016-10-17:
OK, I'll look into this - here are some initial thoughts. The first phase is a utility to run on a PC. It will be a Python3 command line program accepting a path to a .ttf or .otf file, and arguments to define the target row height in pixels and the output format (horizontal or vertical mapping). Its output will be a Python3 source file containing a single font at the specified size and capable of being frozen as bytecode such that font data remains in flash when the module is imported. I'll initially support vertical mapping as this seems most widely used.

It will use Python3 with the Freetype Python bindings.  Fonts will comprise the ASCII character set (chr(32) .. chr(126)) and will be stored as one bit per pixel in a bytes object. To conserve space the character width will be variable with indirect addressing via a lookup table also stored as a bytes object. This is along the lines of the scheme used in the [tft library](https://github.com/peterhinch/micropython-tft-gui.git) - see TFTfont.py and (say) font14.py - the latter being designed for freezing as bytecode.

The second phase would be a Python module to import such a font file and render the font to an arbitrary location in the framebuf. This needs to handle strings honouring (as a minimum) CR characters while tracking the insertion point. Rendering information is probably best handled as a rendering context which remains in place for successive strings until changed by the user. This defines monospaced or variable pitch rendering, plus colour information where the framebuf has more than one bit per character as per tft.py.

kamikaze, 2016-10-17:
@peterhinch could generating and compiling C file save even more space?

peterhinch, 2016-10-18:
In terms of space saving it depends whether you are considering RAM or ROM. RAM is the commodity in short supply. Font files stored as bytes objects and frozen as bytecode use very little RAM indeed when imported. Python font files are typically about 8-12KiB in size, although an unusually large font would be bigger. In terms of flash filesystem storage this is not excessive in my view.

A C solution would require users to edit makefiles to specify the files. It would use some flash (albeit less) in the form of a larger firmware binary. However on many platforms that wouldn't impact the filesystem space. The saving there would be the one advantage on platforms such as the Pyboard with no SD card fitted.

The general approach has been demonstrated on two projects. It is simple to implement: user runs utility, puts output file into frozen modules directory and rebuilds. So my inclination is to stick with a Python solution but I'm open to being convinced otherwise.

peterhinch, 2016-10-22:
I've posted a specification / preliminary user docs [here](https://github.com/peterhinch/micropython-font-to-py). I'd be grateful for any comments or suggestions before I start coding, especially from developers and users of the frame buffer.

mkarliner, 2016-10-22:
That looks pretty good to me Peter, looking forward to it.
While we're talking about framebuffer, the ability to change
the alignment of text (left, center, right), in a given pixel area
would be much appreciated.

peterhinch, 2016-10-22:
@mkarliner Alignment is, of course, a function for the display driver, along with maintaining the insertion point and handling cropping/wrapping and special characters like CR. Currently my principal concern is to identify any issues affecting the format of the generated Python file. Once we have a working utility I'll write a spec for the framebuffer display driver.

Text alignment requires a means of computing the x coordinate where the rendering should commence. This implies a way to determine the length in pixels of a given string, achievable by calling ``get_ch()`` for each character in the string. We implemented centred text in this fashion on the touch LCD GUI.

deshipu, 2016-10-22:
By the way, I think you can write such an utility very easily using PyGame -- it has support for rendering FreeType fonts (and also binary font formats, which would actually be much more useful), and for reading the pixel values of the rendered images -- all you really need for this.

peterhinch, 2016-10-25:
Why do you think binary font formats are more useful? Which specific font file formats are you referring to? I'm not sure what you mean by a FreeType font. FreeType, via the Python bindings, provides a means of converting industry standard ttf and otf files to a one bit per pixel binary format which I then convert to Python source for freezing as bytecode.

Incidentally are there any code samples I could look at for writing to the framebuffer? And are there any actual hardware drivers which use it? These would be of interest when I start on the driver for rendering Python fonts to the framebuf.

mkarliner, 2016-10-25:
ssd1306 uses it.

deshipu, 2016-10-25:
Binary fonts usually contain letters that have been hand- drawn in the particular size pixel-by-pixel and thetefore can be much smaller while still being legible. This is important for low-resolution displays.

As for rendering to framebuffer, you might want to take a look at the pull #2364 where I propose to add a blit method for quickly rendering one buffer on another. This could be useful for text rendering, otherwise you'd have to use the slow 'pixel' method.

deshipu, 2016-10-25:
You can see some examples of binary fonts at http://rasher.dk/rockbox/fonts/jmk/modd-06x11-bold-L1.fnt

deshipu, 2016-10-25:
Sorry, that was http://rasher.dk/rockbox/fonts/jmk/

peterhinch, 2016-10-26:
@mkarliner Thanks for that, I'll get one of those displays for testing.
@deshipu I take your point about binary fonts, but I think that's a separate project which I may well tackle after this one. Converting standard fonts to Python source has been proven on 2.7" e-paper displays and 4.3" LCD's, but I accept it will be below par on very small devices. Where suitable my approach has the advantage of generality, supporting fixed and variable pitch, horizontal and vertical mappings and normal/reversed bit ordering.
Re bitmapped font files, is there any kind of standard with regard to the above parameters or are they designed for specific types of hardware?

peterhinch, 2016-10-28:
Re framebuf mapping, in comments around 4th Feb above there was discussion with @dpgeorge on this subject. The outcome as I understood it was that framebufs would be capable of supporting horizontal or vertical mapping. Looking at the test framebuf1.py and results, and ssd1306.py, it seems currently to support a mode which is a hybrid. The only way I can interpret the data is if bytes are vertically mapped, but subsequent bytes represent the following columns. (In vertical mapping subsequent bytes represent following rows.) The existing mapping is presumably ssd1306 specific.

Is it the intention that framebuf will be general purpose, supporting a variety of mappings? If so, has it been determined what they are to be? This issue has implications for any generic framebuf drawing methods or functions.

deshipu, 2016-10-28:
The patch I linked above adds a `format` argument to specify the format of the framebuffer. It also makes it trivial to add any formats we need. We just need to agree on what those are and how to name them.

peterhinch, 2016-10-28:
OK. But where does the mapping occur? Say you want to provide a routine for drawing a particular shape. Does this have to support every possible mapping? Or can it assume that the framebuf's pixels have a single mapping (or two: horizontal and vertical), with any further translation occurring in the hardware specific device driver? I can see it getting unwieldy and eternally subject to change if there are too many.

deshipu, 2016-10-28:
It happens in the getpixel/setpixel routines. It can't happpen in the device drivers, because the spi and i2c write methods require an object with a buffer protocol (basically a continuous chunk of memory) for no reason, so any remapping would need to double the memory allocation of the framebuffer.

peterhinch, 2016-10-29:
The purpose of a framebuf is to support a set of graphics primitives to avoid these having to be rewritten each time a new hardware device comes along. The reason for supporting more than one mapping is to reduce expensive real time bit twiddling in the hardware device driver. Adding a mapping is only justifiable if it makes a significant real time saving. Its cost is that every graphics primitive needs to be amended and tested against the new hardware.

On this basis I consider that the original proposal of supporting only horizontal and vertical mapping is probably optimal. The writer of (say) a line draw primitive has only to consider two cases. These are mathematically symmetrical: having written and tested one, the other is a simple transformation. And hardware testing requires only two reference platforms.

To justify this I'd like to outline the work required to render a bitmapped object such as a font glyph from a framebuf with vertical mapping to a similarly mapped hardware device, then show that the added work in a device driver to support an SSD1306 mapping is trivial to code and adds nothing to the runtime.

Consider a column of the bitmapped source. It needs to be rendered to an arbitrary row and column defined by bit numbers on the device bitmap. Here is some pseudocode - fb is short for framebuf.

```
fb_offset = offset_into_fb(column_no) # Byte offset
source_offset = offset_into_source(column_no) # byte offset
saved_bits = 0
bit_offset = number of bits each source byte must be right shifted (0..7)
for each byte in the source column
    # clear down bits to be overlain
    if first byte
        fb[fb_offset] &= 0xff >> bit_offset
    elif last byte
        fb[fb_offset] &= 0xff << (8 - bit_offset)
    else
        fb[fb_offset] = 0
    source_byte = source_data[source_offset] >> bit_offset
    source_byte |= saved_bits
    fb[fb_offset] |= source_byte
    saved_bits =  (source_data[source_offset] << (8 - bit_offset)) & 0xff
    source_offset += 1
    fb_offset += 1
```
This is off the top of my head at 7am so it's probably buggy but it gives a flavour of what's involved. In the case of the SSD1306 it is only the last instruction which needs to be modified. If my understanding of the mapping is correct, instead of adding 1 we add the number of columns on the physical device, adding nothing to the runtime.

While it's impossible to anticipate all possible hardware devices I do think there's a case to be argued for supporting two mappings only, neither of which correspond to the one we actually have.

peterhinch, 2016-10-31:
@deshipu Further to the above I have produced some code which I can submit as a PR. I have altered the way the SSD1306 is configured so that it supports true vertical mapping as discussed early in this thread. This is a minor mod to ssd1306.py. I have more extensively altered modframebuf.c to remove the stride argument and to use true vertical mapping.

Vertical and horizontal scrolling work, with areas revealed by the motion blanked. Text rendering works in normal and reverse modes.

In my view this makes the framebuf1 class more generic rather being designed around one specific and rather eccentric mapping of the SSD1306. I anticipate writing a separate framebuf1h class for horizontal mapping. That way, any graphics primitives can be written against exactly two specific generic classes. I think this restriction is important if the framebuf is to be generally useful, but what do you and @dpgeorge think?

If my general approach meets with approval I'll proceed with the PR and with my efforts at rendering arbitrary fonts from Python frozen bytecode.

peterhinch, 2016-11-01:
tl;dr modframebuf.c is somewhat specific to the SSD1306. Restoring the originally intended generality would require a substantial rewrite and would increase the code size by perhaps a factor of 3. My question to the maintainers is whether this is acceptable and whether the effort is justified. I await a response.

As I see it there are 4 options:
 1. Leave it as-is, supporting only the existing nonstandard mapping. Fix known bugs.
 2. Modify it and trivially modify ssd1306.py to support vertical mapping. I have a version which does this.
 3. Enhance (2) to also support horizontal mapping.
 4. Review modframebuf.c to enhance functionality, performance and generality.

The principal reason for (4) is to provide a generic bit blitting facility. The current module supports bidirectional horizontal and vertical scrolling of the entire screen. This is a subset of scrolling a rectangular window, which is itself a subset of bit blitting, where the source data may be from a different array such as a bitmapped image or font glyph. The method must also support the case where source and destination are overlapping regions of the same array as in scrolling. To be general it would need to support horizontal and vertical mapping.

A bit blit would make simplifying the existing text() function desirable. Further the fill() method should be enhanced to support arbitrary rectangular regions. This would aid the case where a scroll using bit blitting needs to clear the region of the display occupied by pixels whose contents have been moved elsewhere.

A bit blit exposed to Python would greatly enhance the performance of device drivers written in Python when rendering bitmapped images including arbitrary font glyphs.

In short this approaches a complete re-write and a significant increase in code size. My concern is whether the idea of a generic framebuf class is actually still of interest to anyone. This started in a plan to write a framebuf driver as a quick way to prove an open source solution to producing fonts. It's raised the above issues; I'm prepared to attempt the task if there is any interest and if such a PR is likely to be accepted.

mkarliner, 2016-11-01:
I'd vote a ++++1 for a generic driver. For example, the SH1106, which frequent turns up as a 1.3 inch oled display, is almost but not quite useable with the existing framebuf. Also, without bit blitting and only single pixel ops the current driver is too slow to do anything very interesting.

deshipu, 2016-11-01:
@peterhinch and @mkarliner what do you think about the changes in #2364? They not only allow you to use 16-bit framebuffers, but also make room for a variety of different formats (without inflating the code size too much, you only need to add two functions for each format) and add a blit method.

peterhinch, 2016-11-01:
@deshipu It looks like a great extension to hardware having its own page buffer.

A few points. As far as I can see the stride variable is only required for SSD1306 devices when configured in their idiosyncratic "horizontal" mode. If configured in vertical mode the need for it disappears. I think there's a strong case for having only two mappings - true horizontal and vertical - as described above. Having variable strides complicates graphics primitives. Is there any actual need for it? Do any display devices actually **require** this curious mapping?

Secondly a bit blit capable of handling overlapping regions of the same buffer reduces the scroll method to a trivial wrapper. Would this not be a useful generalisation and simplification?

I note everything is done on a pixel by pixel basis rather than shifting bytes around. While this is easier to code, is it fast enough?

Lastly, given a bit blit is it not simpler to use it to render characters?

These are details - you've evidently taken this a lot further than my quick hack of SSD1306 based on the code in micropython master.

deshipu, 2016-11-01:
The stride is very useful for when you want to have a "sub-framebuffer" -- that is, a second framebuffer object that shares the buffer with a larger framebuffer, but covers a smaller area. That is very useful for handling windows, scrolling only part of the screen, etc. It's not related to the way pixels are mapped to bits in the bytes of framebuffer.

A blit capable of handling overlapping regions is not a saving in complexity -- that function would be quite complex, in fact. You would just be shifting the complexity around. I think it's better to have specialized functions, so you can use simple and fast code most of the time.

The pixel manipulation done in C is quite fast -- certainly much faster than sending the resulting buffer over to the device over I2C or SPI. Since we are dealing with low resoltions here (due to lack of memory), speed is not a problem. If you look at the history of that pull request, you will note that initially I had specialized code for doing everything efficiently with whole bytes -- but that code grew a lot, and we are more constrained for memory than for speed, so I think it's a good trade-off.

Finally, yes, blitting is the better way to display text -- in fact, I think that the "text" method should be completely removed from the framebuf module, and live in a separate module, dedicated to text processing, with functions for scrolling, line breaking, inverse text, custom fonts, etc. With the blit method being exposed, perhaps that could even be implemented in Python and be fast enough. I left it in there to not change too much at a time.

peterhinch, 2016-11-01:
> It's not related to the way pixels are mapped to bits in the bytes of framebuffer.

Well, in the case of the current SSD1306 driver it is. But I hadn't appreciated the other uses for stride. In terms of mappings of bits into bytes, what mappings do you plan to support?

> that function would be quite complex

Indeed. It's your call, but to some extent scrolling and bit blitting are similar operations and I wondered if there might be a way to exploit that commonality.

>the "text" method should be completely removed from the framebuf module, 
and live in a separate module...

Agree completely. Ideally implemented in Python...

> perhaps that could even be implemented in Python and be fast enough

This is one of the things I started out aiming to test. Having got a font stored as frozen bytecode, can it be rendered to a framebuf at an adequate rate in Python? The drivers I've been worked on don't really test this, the e-paper one using fundamentally slow hardware and the TFT driver having a hardware framebuffer with fast parallel I/O. It is fast, but unrepresentative.
I'll play with my modified SSD1306 driver but the display is too small for results to be meaningful. It's maybe easiest to benchmark with a framebuf not connected to any specific hardware. My interest is to determine whether rendering arbitrary fonts to a framebuf can be achieved in Python in "real time".

dpgeorge, 2016-11-03:
The framebuf module should support (at least) 1-bit displays in both horiz and vert mode, as well as 16-bit displays (with unspecified ordering of the RGB bits).  If we could work in a generic bit blit function then that would be desirable, to render arbitrary glyphs (on 1-bit and 16-bit framebufs) as well as scrolling.

deshipu, 2016-11-03:
The way that I see it, if you ignore flipping and rotating, there are only two 1-bit modes: the one where the least significant bit is first, and the one where it is last. If we do want to include flipping and rotation directly in the format, that adds additional 2**3 models, for a total of 2**4=16 modes. (I think we could have flipping and rotation of the display handled in the driver, though.)

peterhinch, 2016-11-03:
Bit flipping is best handled in the driver, but there are two modes as described in various posts above. These are horizontal mode, where successive bytes represent successive columns of data in a row, and vertical mode where successive bytes represent a column of data. Doing this transformation in the driver is rather heavyweight.

See the post from @dpgeorge on 4th Feb.

Assume an array of pixels p[x, y]. This is stored in bytes. In horizontal mode, the bits of the first byte represent (MSB first)
p[0, 0], p[1, 0] .. p[7, 0] and the second byte is p[8, 0] .. p[15, 0]
In vertical mode we have (LSB first)
p[0, 0], p[0, 1] .. p[0, 7] and the second byte [0, 8] .. p[0, 15]

Alas the "horizontal" mode of the SSD1306 is neither of these, although the vertical mode is correct. Which is why I advocate changing this driver.

deshipu, 2016-11-03:
What happens to your "horizontal" and "vertical" modes when you rotate your display 90°?

deshipu, 2016-11-03:
We can include flipping/rotation in the framebuffer itself, it's a rather simple function to be applied to all coordinates before using them.

peterhinch, 2016-11-03:
You and @dpgeorge are the experts on this - I await your conclusions with interest. I took it that realtime rotation would be expensive in memory accesses and processor cycles. But supporting a single mapping has obvious advantages.

deshipu, 2016-11-03:
The kind of rotation I mean happens to the coordinates, not to the actual pixels -- so the buffer stays how it was, but any new drawing operations will use the new coordinate system. This would be usually just subtracting something, or swapping two coordinates.

peterhinch, 2016-11-03:
OK, we're at cross purposes. I'm considering the relationship between the pixel to byte mapping of the buffer as compared to that of the physical device. This was what the discussion I referenced was about, with e-paper exemplifying horizontal mapping and the Pyboard LCD shield exemplifying vertical (the SSD1306 too if configured that way).

deshipu, 2016-11-03:
The thing is, by configuring the flipping and rotation for your display, you can cover most of the "modes" (that is, decide where the (0, 0) pixel is on your screen) -- the only change you can't control is the LSB/MSB one, so that will need to stay with the format. Or am I missing something?

peterhinch, 2016-11-04:
There's a clear benefit when it comes to transferring data to the device, if the mapping of bits onto bytes corresponds with the mapping on the physical device: you can do bytewise transfers or block transfers if the device supports them. Hence the proposal to support true horizontal and vertical mapping as these are common on physical devices.

I think we're saying the same thing in different ways. I was arguing that these two mappings can be supported because, from the point of view of someone writing (say) a line drawing primitive, there is a simple mathematical transformation between the two. Which is what you're saying above. In other words there are two closely related test cases. Extra mappings, like the current weird SSD1306 one, should be avoided firstly because the transformation is not so simple, and secondly because each mapping adds a test case to every graphics primitive.

My suggestion is that the SSD1306 driver should be changed to support vertical mapping, and that mappings other than vertical or horizontal should be strongly deprecated.

deshipu, 2016-11-04:
I think I need to use pictures. Consider this:

[image]

This what the SSD1306 format looks like. Each byte is a column of 8 pixels, arranged starting from the most significant one to the least significant bit. That's what I assume you mean by "vertical mapping". Now, if you rotate the physical pixel by 90°, you will get this:

[image]

Here each byte is a row of 8 pixels, also starting from the most significant bit. That's the "horizontal mapping". Now lets try flipping. If I flip this image horizontally, I will get:

[image]

Here each byte is a row of 8 pixels, but starting from the least significant bit. If I rotate that 90° again, I get:

[image]

Where each byte is a column of pixels again, starting with the least significant bit.

That covers all sane mappins from bytes to pixels. However, it doesn't cover all possible places where the (0, 0) coordinate is -- for that you need all the remaining flips and rotations. But in the end, there is only one "format", it's just rotated and/or flipped around.

deshipu, 2016-11-04:
So generally, there are 4 parameters:

- the direction of the x coordinate (left or right)
- the direction of the y coordinate (up or down)
- the swapping of the x and y coordinates (whether the bytes are vertical or horizontal)
- the direction of bits in a byte (lsb or msb first)

deshipu, 2016-11-04:
The first three can be handled with the following function called on all coordinates before using them:

```
void flipxy(byte flip, int w, int h, int &x, int &y) {
    if (flip & 0x04) {
        int tmp = x;
        x = y;
        y = tmp;
        tmp = w;
        w = h;
        h = tmp;
    } 
    if (flip & 0x01) {
        x = w - x;
    }
    if (flip & 0x02) {
        y = h - y;
    }
}
```

[9 more replies]

closed 2024-07-20 by mattytrentini