micropython/micropython #1818: native code emitter: bug or known limitation?

peterhinch, 2016-02-06:
```python
import array
@micropython.native
def foo():
    a = array.array('i',(pyb.rng() for _ in range(10)))
```
Produces:
```
MicroPython v1.6-10-gff1a96c on 2016-02-04; PYBv1.0 with STM32F405RG
[...]
>>> foo()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
MemoryError: memory allocation failed, allocating 2592941848 bytes
>>> 
```

pfalcon, 2016-02-06:
Generators aren't supported with @native: https://github.com/micropython/micropython/issues/1041

dpgeorge, 2016-02-06:
On unix an assertion (should) fail.  On pyboard the assertions are turned off so that's why it tries to run by gives a strange error.

peterhinch, 2016-02-07:
OK, thanks for that. I'll put a note to that effect in my forthcoming doc.

closed 2016-02-10 by dpgeorge