micropython/micropython #1824: There is no header file for ESP8266

choies1, 2016-02-08:
I tried to build MycroPython for ESP8266. However, I met the following error.

**main.c:39:28: fatal error: user_interface.h: No such file or directory**

where can I find 'user_interface.h'.

Thanks.
-----------------------------------------------------------------
$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build/genhdr/qstrdefs.generated.h
Generating build/genhdr/mpversion.h
mkdir -p build/build
mkdir -p build/lib/libc
mkdir -p build/lib/mp-readline
mkdir -p build/lib/netutils
mkdir -p build/lib/timeutils
mkdir -p build/lib/utils
mkdir -p build/py
mkdir -p build/py/../extmod
mkdir -p build/stmhal
CC ../py/mpstate.c
CC ../py/nlrx86.S
CC ../py/nlrx64.S
CC ../py/nlrthumb.S
CC ../py/nlrxtensa.S
CC ../py/nlrsetjmp.c
CC ../py/malloc.c
CC ../py/gc.c
CC ../py/qstr.c
CC ../py/vstr.c
CC ../py/mpprint.c
CC ../py/unicode.c
CC ../py/mpz.c
CC ../py/lexer.c
CC ../py/lexerstr.c
CC ../py/lexerunix.c
CC ../py/parse.c
CC ../py/scope.c
CC ../py/compile.c
CC ../py/emitcommon.c
CC ../py/emitbc.c
CC ../py/asmx64.c
CC ../py/emitnative.c
CC ../py/asmx86.c
CC ../py/emitnative.c
CC ../py/asmthumb.c
CC ../py/emitnative.c
CC ../py/emitinlinethumb.c
CC ../py/asmarm.c
CC ../py/emitnative.c
CC ../py/formatfloat.c
CC ../py/parsenumbase.c
CC ../py/parsenum.c
CC ../py/emitglue.c
CC ../py/runtime.c
CC ../py/nativeglue.c
CC ../py/stackctrl.c
CC ../py/argcheck.c
CC ../py/warning.c
CC ../py/map.c
CC ../py/obj.c
CC ../py/objarray.c
CC ../py/objattrtuple.c
CC ../py/objbool.c
CC ../py/objboundmeth.c
CC ../py/objcell.c
CC ../py/objclosure.c
CC ../py/objcomplex.c
CC ../py/objdict.c
CC ../py/objenumerate.c
CC ../py/objexcept.c
CC ../py/objfilter.c
CC ../py/objfloat.c
CC ../py/objfun.c
CC ../py/objgenerator.c
CC ../py/objgetitemiter.c
CC ../py/objint.c
CC ../py/objint_longlong.c
CC ../py/objint_mpz.c
CC ../py/objlist.c
CC ../py/objmap.c
CC ../py/objmodule.c
CC ../py/objobject.c
CC ../py/objpolyiter.c
CC ../py/objproperty.c
CC ../py/objnone.c
CC ../py/objnamedtuple.c
CC ../py/objrange.c
CC ../py/objreversed.c
CC ../py/objset.c
CC ../py/objsingleton.c
CC ../py/objslice.c
CC ../py/objstr.c
CC ../py/objstrunicode.c
CC ../py/objstringio.c
CC ../py/objtuple.c
CC ../py/objtype.c
CC ../py/objzip.c
CC ../py/opmethods.c
CC ../py/sequence.c
CC ../py/stream.c
CC ../py/binary.c
CC ../py/builtinimport.c
CC ../py/builtinevex.c
CC ../py/modarray.c
CC ../py/modbuiltins.c
CC ../py/modcollections.c
CC ../py/modgc.c
CC ../py/modio.c
CC ../py/modmath.c
CC ../py/modcmath.c
CC ../py/modmicropython.c
CC ../py/modstruct.c
CC ../py/modsys.c
CC ../py/vm.c
CC ../py/bc.c
CC ../py/showbc.c
CC ../py/repl.c
CC ../py/smallint.c
CC ../py/frozenmod.c
CC ../py/../extmod/moductypes.c
CC ../py/../extmod/modujson.c
CC ../py/../extmod/modure.c
CC ../py/../extmod/moduzlib.c
CC ../py/../extmod/moduheapq.c
CC ../py/../extmod/moduhashlib.c
CC ../py/../extmod/modubinascii.c
CC ../py/../extmod/machine_mem.c
CC ../py/../extmod/modussl.c
CC ../py/../extmod/modurandom.c
CC ../py/../extmod/fsusermount.c
CC ../py/../extmod/moduos_dupterm.c
CC strtoll.c
CC main.c
main.c:39:28: fatal error: user_interface.h: No such file or directory
compilation terminated.
make: *** [build/main.o] Error 1

alex-march, 2016-02-08:
The ESP8266 port relies on the [esp-open-sdk](https://github.com/pfalcon/esp-open-sdk/) to be installed and xtensa-* tools available in user PATH. In the SDK there is a bunch of vendor defined APIs, such as "user_interface.h" which is not found in your case, and they are being used by the micropython port. Refer to the installation instruction in the esp-open-sdk repository. Once you have the tools ready try running make again and everything should work!

bhuvanchandra, 2016-02-09:
As @alex-march mentioned, after setting up the [esp-open-sdk](https://github.com/pfalcon/esp-open-sdk/), export the toolchain path and then build MicroPython. Lately this information is added in ESP8266 readme file also.

closed 2016-02-10 by dpgeorge