micropython/micropython #1825: Usart 1 will not initialize from main.py

neuralauv, 2016-02-08:
Have attempted to initialize USART1 in main.py as follows:

Device1=UART(1,4800)
Device2=UART(6,9600)

from REPL if I then say:
a=Device1.readline()  I get nothing in a
if I redo the experiment with the device plugged into 6, a has data.
Then if I manually retype from REPL
Device1=UART(1,4800)
and a=Device1.readline() 
a has valid data

dpgeorge, 2016-02-08:
I don't understand the problem.

> from REPL if I then say:
> a=Device1.readline() I get nothing in a

Why do you expect to have something in a?  Is there anything physically connected to UART1?

> if I redo the experiment with the device plugged into 6, a has data.

Please describe what "redo the experiment" means, and what "the device" is and what "6" is referring to.

> Then if I manually retype from REPL
> Device1=UART(1,4800)
> and a=Device1.readline()
> a has valid data

What exactly does a have in it?

neuralauv, 2016-02-09:
Hi Damien,
to be more direct. I have two peripherals, an EM406 GPS and an xbee 
series one radio. Since the gps sends data updates on its own it is a 
good test of the unit. 
 
my main .py program had this intialization:
 
from pyb import UART
 
gps=UART(6,4800)
then from the command line I only need to type
gps.readline() to see data. same if I plug the xbee radio into port 6 
with baud rate set apporpriately
 
the way I have the carrier board, the next port connection is port 1. 
 
IF I plug the gps into port 1, and adjust the code to read:
 
from pyb import UART
gps=UART(1,4800)
 
then reboot and try
 
gps.readline().. I get nothing....but
 
at this point, if I manually type gps=UART(1,4800)
then type gps.readline()   it works fine. 
 
Just to check I also connected to port 4, and adjusted the code in 
main.py and it worked fine just like port 6. 
 
The radio does the same thing. So it seems to be something specific to 
the port 1 initialization. I manually tried to set the alternate 
functions to pushpull and the correct alternate function
in main.py and it didn;t alleviate the problem. 
 
any help is appreciated. 
 
thanks,
Brian Howell
 
if instead
On Mon, 08 Feb 2016 15:03:55 -0800, Damien George  wrote:

       I don't understand the problem. 

       from REPL if I then say:
a=Device1.readline() I get nothing in a

Why do you expect to have something in a? Is there anything physically 
connected to UART1?

       if I redo the experiment with the device plugged into 6, a has data. 

Please describe what "redo the experiment" means, and what "the device" 
is and what "6" is referring to. 

       Then if I manually retype from REPL
Device1=UART(1,4800)
and a=Device1.readline()
a has valid data

What exactly does a have in it?
—

   *

dpgeorge, 2016-02-09:
I cannot reproduce the bug.  Here is what I did:

- make pyboard 1 spit out a string every second on one of its UARTs at 4800 baud
- get pyboard 2 and put the following in its main.py:
```
from pyb import UART
ua = UART(1, 4800)
```
- connect UART of pyboard 1 to UART(1) of pyboard 2
- hard reset pyboard 2
- access REPL of pyboard 2
- type `ua.readline()`
- result from the readline call is that I get the string sent from pyboard 1, as expected
- if I do a soft reset (ctrl-D) on pyboard 2 then main.py is executed again and `ua.readline()` still works and returns the string from pyboard 1 (which is still spitting them out)

@neuralauv do you have anything in boot.py or main.py that is modifying the pins or UART(1) somehow?

JonHylands, 2016-02-09:
Just a reminder - a slow GPS that runs at 4800 baud probably only updates
at 1 Hz, so if your gps.readline() happens only once immediately after you
create the UART, then it probably won't return anything. Doing it on the
command line manually probably gives you that 1+ second delay.

If that's the case, try adding a 1 second delay before you call
gps.readline(), or call it in a loop that runs for more than one second.

peterhinch, 2016-02-09:
I've posted a suggestion in the forum of testing using a loopback. For what it's worth I can't replicate this fault, either.

neuralauv, 2016-02-09:
Hi damien,
Thanks for the response. 
I have nothing additional in boot.py, and the code I described resides 
in main.py. 
I did one other thing last night and encapsulated the gps 
initialization and read into a function as so:
 
def gps_read(port):
    from pyb import UART
    gps=UART(port,4800)
    a=gps.readline()
    return a
 
and put it in main.py. 
 
Then if I call it from repl, it works fine. 
 
One quick question. These are all older boards for which I have 
upgraded the dfu image.  For your experiments,
Which firmware image are you using?
 
I shall repeat your experiment today. 
 
cheers,
Brian

On Tue, 09 Feb 2016 03:37:03 -0800, Damien George  wrote:

       I cannot reproduce the bug. Here is what I did:

   * make pyboard 1 spit out a string every second on one of its UARTs 
at 4800 baud
   * get pyboard 2 and put the following in its main.py:

from pyb import UART
ua = UART(1, 4800)

   * connect UART of pyboard 1 to UART(1) of pyboard 2
   * hard reset pyboard 2
   * access REPL of pyboard 2
   * type ua.readline()
   * result from the readline call is that I get the string sent from 
pyboard 1, as expected
   * if I do a soft reset (ctrl-D) on pyboard 2 then main.py is 
executed again and ua.readline() still works and returns the string 
from pyboard 1 (which is still spitting them out)

@neuralauv do you have anything in boot.py or main.py that is modifying 
the pins or UART(1) somehow?
—

   *

neuralauv, 2016-02-09:
Hi Jon,
Thanks for that reminder, yes it can be a factor if I suck the buffer 
dry. In this case on my setup, I found the same problem with my radio 
chip when attaching to port one. 
PLEase see my comments to Damien on my work around, but in addition if 
you could let me know which firmware version you are utilizing for your 
experiments, I will duplicate that on this end. 
 
cheers,
Brian

On Tue, 09 Feb 2016 03:56:31 -0800, Jon Hylands  wrote:

       Just a reminder - a slow GPS that runs at 4800 baud probably 
only updates
at 1 Hz, so if your gps.readline() happens only once immediately after you
create the UART, then it probably won't return anything. Doing it on the
command line manually probably gives you that 1+ second delay. 

If that's the case, try adding a 1 second delay before you call
gps.readline(), or call it in a loop that runs for more than one second. 

—

   *

neuralauv, 2016-02-09:
Thanks Peter,
Will do
 
Brian

On Tue, 09 Feb 2016 05:29:51 -0800, Peter Hinch  wrote:

       I've posted a suggestion in the forum of testing using a 
loopback. For what it's worth I can't replicate this fault, either. 
—

   *

dpgeorge, 2016-02-09:
I'm using PYBv1.0 with v1.6 firmware.

neuralauv, 2016-02-09:
Ok will duplicate on my set up

On Tue, 09 Feb 2016 06:02:34 -0800, Damien George  wrote:

       I'm using PYBv1.0 with v1.6 firmware. 
—

   *

dpgeorge, 2016-02-15:
@neuralauv is this now resolved?

neuralauv, 2016-02-15:
Hi Damien,
I have not had an opportunity to verify the operation. But I have 
synchronized all my upy boards to 1.6 so will be testing tomorrow. 
 
Thanks
Brian

On Mon, 15 Feb 2016 00:45:16 -0800, Damien George  wrote:

       @neuralauv is this now resolved?
—

   *

dpgeorge, 2016-02-15:
@neuralauv ok, I just saw there was progress on the forum at http://forum.micropython.org/viewtopic.php?f=2&t=1367 and thought it was resolved.  What hardware/board are you running that the error occurs on?

dhylands, 2016-02-15:
I believe that @neuralauv is using a 407 Discovery board.

With https://github.com/micropython/micropython/commit/ec37239e53262a97c514f8948b3edc73dcfe1188 landing, UART 1 will no longer be defined on the 407 Discovery board (since it isn't usable anyways).

UARTs 2, 3, and 6 should all be usable on the discovery and uart 4 if you disable the user switch.

neuralauv, 2016-02-17:
Hi Damien,
I got to work the issue today. Attempting to duplicate your set up. 
With two pyboards connected on port 1 I did have similar issues, but 
when I initialized the "read_buf_len" variable on the UART, everything 
worked well. 
 
I then went back to my project hardware and achieved the same result, 
so the issue was peculiar to port 1 requiring I intitalize the read 
buffer. All is well now so go ahead and close out the ticket. 
 
Much thanks to you and your team!
 
cheers,
 
Brian

On Mon, 15 Feb 2016 10:27:16 -0800, Damien George  wrote:

       @neuralauv ok, I just saw there was progress on the forum at 
http://forum.micropython.org/viewtopic.php?f=2&t=1367 and thought it 
was resolved. What hardware/board are you running that the error occurs 
on?
—

   *

dpgeorge, 2016-02-17:
Ok, thanks @neuralauv .

closed 2016-02-17 by dpgeorge