micropython/micropython #1828: docs RFC on speed optimisation guide

peterhinch, 2016-02-09:
I've made a first pass on this [here](http://hinch.me.uk/html/reference/speed_python.html). I'm looking for comments on the general approach, level of detail and things I might have left out - especially for any factual errors and omissions.

@dpgeorge my knowledge of Native and Viper isn't all it might be; there are two specific cries for help in the Viper section!

dpgeorge, 2016-02-09:
Problem with viper ptr casting should be fixed in 3e02b1d19a4b1fb36ee36b0d43144bad6b797f2f.  Thanks for finding the bug!

peterhinch, 2016-02-09:
Ah, we're chasing each others tails here :) In working round this I found #1829.

peterhinch, 2016-02-09:
On the topic of docs I'd like to include a list of operations resulting in heap allocation. Those I'm aware of are:
* float operations (and, by implication, complex).
* Instantiating classes including user defined classes.
* Instantiating lists, sets or dicts.
* Appending to lists or adding items to sets and dicts (I assume removing items is OK).
* Instantiating long ints and arithmetic/logical operations on them.
* Creating slices of arrays, bytearrays or bytes objects.
* Operations which appear to change immutable objects such as strings or bytes (e.g. s1 += s2).

Doubtless there are others. If you think this is a useful way to go, perhaps you could advise.

dpgeorge, 2016-02-09:
I already have a list/document of things requiring heap allocation, so you can leave that to me :)

ynohtna, 2016-02-10:
This is incredibly useful. Thank you so much, Peter!

peterhinch, 2016-02-15:
Viper section now updated based on forum posts from @dpgeorge. Assorted typos expunged.

peterhinch, 2016-02-18:
I've now submitted a PR for review so I'll close this.

closed 2016-02-18 by peterhinch