micropython/micropython #1829: viper: new ptrx API. Code behaves oddly if pasted at REPL

peterhinch, 2016-02-09:
I was playing with this code by pasting at the REPL (using firmware issued 8th Feb on Pyboard V1). Each time I modified the bytearray length, reducing from a high initial value to 2. The code compiles and runs if I paste it immediately after a soft reset.
```python
from uctypes import bytearray_at
BIT0 = const(1)
@micropython.viper
def toggle_n(n: int):
    odr = ptr16(bytearray_at(stm.GPIOA + stm.GPIO_ODR, 2))
    for _ in range(n):
        odr[0] ^= BIT0
```
If I paste in another version, with only the bytearray length altered, I get
```python
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== @micropython.viper
=== def toggle_n(n: int):
===     odr = ptr16(bytearray_at(stm.GPIOA + stm.GPIO_ODR, 4))
===     for _ in range(n):
===         odr[0] ^= BIT0
=== 
Traceback (most recent call last):
  File "<stdin>", line 4, in toggle_n
ViperTypeError: can't do binary op between 'int' and 'object'
>>> 
```

dpgeorge, 2016-02-09:
This is because the constant BIT0 is not available as a constant in the second version (constants are only available in a single compilation unit).  But BIT0 is still a global variable, so the second version would (should) work if you cast it: `odr[0] ^= int(BIT0)`.

peterhinch, 2016-02-09:
<fx> Light dawns. It does. Thanks.

dpgeorge, 2016-02-09:
So the issue really is: const only optimises across the current compilation unit.  But that's a known limitation.

closed 2016-02-09 by dpgeorge

peterhinch, 2016-02-09:
Entirely reasonable behaviour.