micropython/micropython #1831: Best way to interrupt the python script to call a function from a software interrupt.

israelg99, 2016-02-09:
We have an RTOS where microPython runs on a single thread.
Periodically we want the microPython VM to handle events from other parts of our system.
For instance, asynchronously call a function from C in Python.

Here is an example:
```
def callFromC()
    print(“event from c sub system”)

def main()
    timer.setCallback(callFromC)
    timer.start(500)
            while(1):
                print(“.”)
                sleep(100)
```
 
I am thinking that some place in the microPython VM I could call an event handler that then runs the python callback, but I am not sure the best place for this event handler.
Any ideas?

dpgeorge, 2016-02-09:
This is something that's important to be able to do, so we will very soon be providing a hook in the VM to make such calls.  The place to put it is after this line: https://github.com/micropython/micropython/blob/master/py/vm.c#L1254

Try adding something like:
```c
static uint vm_hook_divisor = 0;
if ((++vm_hook_divisor & (VM_HOOK_DIVISOR - 1)) == 0) {
    vm_hook_function();
}
```
where VM_HOOK_DIVISOR should be a constant that is a power of 2 (eg 16).

PappaPeppar, 2016-02-09:
Is this not the same as software interrupts #1788 but initated from the rtos?

AWCharlie, 2016-02-09:
Please excuse the interruption. One of our guys has figured out the ESP8266
UART on boot with RTS/CTS and all that. If this is new info, where can he
contribute to the ESP8266 project?

-- Charlie Springer

dhylands, 2016-02-09:
> Please excuse the interruption. One of our guys has figured out the ESP8266 UART on boot with RTS/CTS and all that. If this is new info, where can he contribute to the ESP8266 project?

Please don't hijack somebody else's thread and totally change the topic. Instead, find an appropriate thread or create a new one.

AWCharlie, 2016-02-10:
It says "issues". I didn't want to create an issue. Too late I see.

dpgeorge, 2017-06-14:
There is now a MICROPY_VM_HOOK_LOOP macro to hook into the VM, as well as a scheduler to schedule a callback to happen "soon".

closed 2017-06-14 by dpgeorge