micropython/micropython #1836: I2C operations on WiPy do throw OSErrors

kfricke, 2016-02-10:
After non-reproducible timespans during both read and write I2C operations (i do only use readfrom and writeto) the following error is thrown on my WiPy:
`OSError: the requested operation failed`

This behavior occurs while running drawing tests on my SSD1306 OLED display and while communicating with my environmental sensors (so far both MCP9808 and HDC1008).

While trying to tackle this issue i did deactivate interrupts before starting the I2C communication, but this did not solve the issue.
I have not yet looked into memory usage and GC behaviour before and after the I2C operations. Maybe it just throws timeouts during GC!?

It should be noted that those devices do work flawlessly on my Pyboard with the same code. The only differences are in the I2C class of the machine module: `readfrom `instead of `recv` and `writeto` instead of `send`.

The code for those devices is publicly available and i do spare the some code, because this issue do occur in all repeated I2C commands.

danicampora, 2016-02-10:
Hello,

Thanks for the report. 2 questions:

1. What's the bus speed? is it 400KHz?

2. Once it happens the first time, then the exception is raised every single time until a soft-reset?

Cheers,
Daniel

kfricke, 2016-02-10:
Yes, the bus speed was set to 400 kHz, but another test with the speed set to 100kHz did throw the same OSError while typing this comment (this time while reading on the bus).

The I2C object is initialized the following way:
```
from machine import I2C
...
i2c = I2C(0, mode=I2C.MASTER, baudrate=400000, pins=('GP24', 'GP23'))
```
Here is a little transcript of what's happening. The script does dump the values of both sensor to the console every 250 ms continuously...
```
...
MCP9808 - Temp: 22.56
HDC1008 - Temp: 22.70 | Humi: 52.00
MCP9808 - Temp: 22.56
HDC1008 - Temp: 22.70 | Humi: 52.00
Traceback (most recent call last):
  File "main.py", line 44, in <module>
  File "/flash/lib/mcp9808.py", line 103, in get_temp_int
  File "/flash/lib/mcp9808.py", line 39, in _send
OSError: the requested operation failed
MicroPython v1.6-21-g9e78ab4 on 2016-02-09; WiPy with CC3200
Type "help()" for more information.
>>> i2c.scan()
[24, 64]
>>> mcp.get_temp_int()
(22, 81)
>>> hdc.get_temp_humi(int_only=True)
((22, 70), (52, 0))
```
`i2c` is at the context after the crash the I2C object, which the communication did run on. Both `mcp` and `hdc` are objects representing my sensors. So you can see that even without a soft-reset the communication works again.

The error occured on the last line of this method:
```
def _send(self, buf):
        """
        Sends the given buffer object over I2C to the sensor.
        """
        self._i2c.writeto(self._addr, buf)
```

Maybe this is really just some unexpected timeout or so?

BTW: the naming scheme of the methods in the WiPy machine.I2C class does not conform to [PEP 0008 regarding function name syntax](https://www.python.org/dev/peps/pep-0008/#function-names) (`writeto` instead of `write_to`)

kfricke, 2016-02-10:
I do only own a cheap Buspirate as a logic analyzer. So tracing such an issue which occurs sporadically after seconds or minutes is out of my reach until you can give me some good advice.

danicampora, 2016-02-23:
@kfricke could you check is this issue persists on the latest release: https://github.com/wipy/wipy/releases/tag/1.2.0 ? Thanks.

kfricke, 2016-02-28:
Nothing has changed, sorry! Checked with `MicroPython v1.6-89-g440d33a on 2016-02-27; WiPy with CC3200` and `MicroPython v1.6-93-g8a18084 on 2016-02-28; WiPy with CC3200`

dpgeorge, 2020-06-09:
I don't know if this issue still exists but likely it'll never get attention, so I'll close it for now.

closed 2020-06-09 by dpgeorge