micropython/micropython #1838: Currently no way to specify boot.py or main.py using frozen files

dhylands, 2016-02-11:
It would be nice to be able to specify boot.py and/or main.py as frozen files, especially when building for REPL-less devices.

dpgeorge, 2016-03-17:
This feature should be supported, and would be a preferred solution to #1891.

pfalcon, 2016-04-11:
The way to implement this was discussed and settled in https://github.com/micropython/micropython/issues/1955

dhylands, 2016-04-11:
I'm not seeing it in #1955. As far as I can tell it still requires main.py to be from a filesystem and not from a frozen module?

dhylands, 2016-04-11:
I guess, more specifically, I'd like to see main.py come from frozen so that a system with NO filesystem can be booted (useful for some of the lower-memory devices).

peterhinch, 2017-06-20:
Another use case for this is that it enables a firmware image to be distributed where all Python code is frozen. The user merely needs to flash the image, reboot and run.

There are obvious but device specific workrounds on the ESP8266, but freezing main.py does seem the "natural" solution if only it actually worked.

dpgeorge, 2019-05-01:
Support for frozen boot.py/main.py was added in 859596ce25bcaa9b6104f16b8b55d0832c55e3f8

closed 2019-05-01 by dpgeorge